realsenseai/RealSenseID #2: networked realSenseIDs

3dsf, 2021-01-14:
Hi,

If there are multiple points of entry, do you have introduce each user individually to each device?  Or is their some feature to pass profiles between devices?

thx

alexk1976, 2021-01-14:
Hi,
An excellent question.. we are working on this feature and it'll part of our next release in a month. As part of it we'll provide an additional APIs/code to retrieve faceprints from device as well to manage users and matching them on server side. Customers will be able to have centralized server/DB with user faceprints and call our match API to allow/forbid  specific user access.
thanks

3dsf, 2021-01-14:
Great,

I had looked through of what was currently available and didn't see a reference to the capability.  I was afraid the reference to a 1000 profile on device storage and the lack of reference to such a feature on the Architecture Diagram, that it might not be possible.  

Thanks : )

closed 2021-01-14 by 3dsf

jshepin, 2021-01-19:
this feature is extremely important for businesses and buildings in order to use this in an application environment. I'm not sure why intel has not supported this be default.