realsenseai/RealSenseID #5: Linux (Coming soon)

3dsf, 2021-02-08:
The marketing mentions that linux systems will be supported soon; is it expected to be supported at launch?

alexk1976, 2021-02-08:
Linux already supported and was validated on Ubuntu.
See readme on the main page.
thanks

3dsf, 2021-02-09:
Great, thanks.

Sorry, I was asking based off of https://www.intelrealsense.com/facial-authentication/ and I should have consulted the readme here.  

---
**Successfully built RealSenseID 0.11.0 on Fedora 33** 
gcc (GCC) 10.2.1 20201125 (Red Hat 10.2.1-9)
_1 warning_
```
[ 93%] Building CXX object wrappers/c/CMakeFiles/rsid_c.dir/src/rsid_c_client.cc.o
/home/3dsf/git/RealSenseID/wrappers/c/src/rsid_c_device_controller.cc: In function ‘rsid_device_controller* rsid_create_device_controller()’:
/home/3dsf/git/RealSenseID/wrappers/c/src/rsid_c_device_controller.cc:29:28: warning: deleting ‘void*’ is undefined [-Wdelete-incomplete]
   29 |                 delete rv->_impl;
      |                            ^~~~~
```

gabime, 2021-02-09:
Fixed the warning (fe6c4dbba4e5a0db0c1654191fb5f8b2f2641959). Thanks for the notice.

closed 2021-02-09 by gabime