realsenseai/RealSenseID #8: Where is the firmware(F450 FW 1.8.2 F450_FW_1.8.2.7z)?

bonobono999, 2021-02-17:
I have a F455 model.
I received "Intel RealSenseID SDK".

Execute the rsid_c_example.exe program and execute pairing. The following message appears.

What should I do?

[2021-02-17 14:40:26.440] [debug] [WindowsSerial] Opening serial port COM6 type usb
[2021-02-17 14:40:26.453] [info] [FaceAuthenticatorImpl] Pairing start
[2021-02-17 14:40:26.454] [debug] [PacketSender] Sending packet 'h'
[2021-02-17 14:40:26.457] [debug] [PacketSender] Waiting packet..
[2021-02-17 14:40:26.457] [error] [PacketSender] Protocol version doesn't match. Expected: 0, Received: 98
[2021-02-17 14:40:26.458] [error] [FaceAuthenticatorImpl] Failed to recv device ecdsa public key
Error status: 104 (VersionMismatch)

nick-nam, 2021-02-17:
Please contact your Intel RealSense Group account manager.

bonobono999, 2021-02-22:
Upgraded new firmware(1.8.2).

closed 2021-02-22 by bonobono999

rkyymmt, 2021-03-01:
I received preordered RealSenseID F455 today.
But firmware version is old! (1.6.1.1007)
I can't try this SDK.
Please send me new version of firmware.

nick-nam, 2021-03-03:
Please contact your Intel Account Manager and/or request latest SDK via Zendesk if you have an account.