realsenseai/RealSenseID #10: how to store Faceprints-Extraction in server mode and find matches?

nickticket2u, 2021-02-25:
in server mode, I understand that the device can return "Faceprints-Extraction" data.

is there a documentation/explanation on what this data is , and how to store it in a central database?

and when doing matching server mode, do we have to loop every data to check one by one ?

alexk1976, 2021-02-25:
Data is defined in Faceprints.h. it's up to user to decide where/how to store it. 
yes, user is expected to run loop over database till match is succeeded . 
I recommend looking for matching code in  c++ sample though data is stored in runtime memory there and real application should store it in real database.

nickticket2u, 2021-02-25:
thanks alex. really appreciate your response. still waiting for the camera 

I think it's clearer to me now.

Let's say I store 1000 faceprints in a remote database.

Then I start a program on a new computer with new realsense id camera. 

The program has to get all those 1000 faceprints, and enroll them one by one on the camera.

By that way, I can directly use authenticate. no need to loop using MatchFaceprints

This could work right ? Apologies for asking so many questions, even before receiving the camera

alexk1976, 2021-02-25:
you have 2 options: device (database and matching done in device) or server mode (database and matching done in host/server).
In the first case you can all to authenticate and we do matching for you. In the second mode upon authentication you get faceprints and you must match them over database you manage in host.
In the server mode there is no 1000 limitation. You run authenticate whenever you want but in the server mode you must match at some point otherwise you dont know the user id.

nickticket2u, 2021-02-25:
thanks a lot !

closed 2021-02-25 by nickticket2u