realsenseai/RealSenseID #12: Preview is very dark

nickticket2u, 2021-03-08:
Hello people.

Just wonder why the camera's PreviewImage returned is very dark, like almost pitch black. I've switch on all the lights, face a flashlight on my face, it is still extremely dark. I see that it has some bluish tint, maybe that's the reason.

Tried with the c# example. Same with my own project.

It will ID correctly though.

Any help is greatly appreciated

nickticket2u, 2021-03-08:
point to the white air conditioner, click Auth. preview then becomes a bit bright
[image]

Points to a dark place, click Auth. Then point back to myself. Preview then becomes dark

[image]

RealSenseSupport, 2021-03-08:
Hi @nickticket2u 

Thanks for submitting this issue.  We will have a look into this and let you know what we find out.

Just to confirm again, you're not seeing an issue with authenticate even though the preview looks like this?

nickticket2u, 2021-03-08:
thanks for the response.
yeah authenticate works fine.

3dsf, 2021-03-16:
hey @nickticket2u  

I've been watching your ticket, as I recognise value in the preview feature.

Just a heads up, instead of doing edits to the first post, it might be better to make new comments, as they would alert the people involved with this project with the further information related to this issue.  
On a platform such as reddit, the edits in the post work well because you are counting on continual new viewers.

ShravyaB, 2021-03-25:
I have observed an opposite behaviour. I was testing a backlight situation but this occurs in a well lit situation too. When a person enters the frame (even if it is far) there is higher exposure compensation done (light in the background gets bloated). Is this done on purpose?

3dsf, 2021-04-02:
@nickticket2u

Does RealSenseID/releases/tag/v0.15.0 resolve the issue for you ?

nickticket2u, 2021-04-03:
Will have time to try it next week 🙏😁

nickticket2u, 2021-04-07:
problem is solved with latest firmware

closed 2021-04-07 by nickticket2u