realsenseai/RealSenseID #15: Publish API Reference on Web

UnaNancyOwen, 2021-03-13:
API Reference for RealSense ID SDK (generated by doxygen) is not publicly available.  
Please publish API Reference on Web like [RealSense SDK](https://intelrealsense.github.io/librealsense/doxygen/index.html).  
I'm temporarily publish API Reference on Web, but it is unofficial.
By all rights, It should be publish officially by RealSense ID team. Thanks,
https://unanancyowen.github.io/realsenseid_apireference/

gabime, 2021-03-13:
The doxygen is available in the release page (https://github.com/IntelRealSense/RealSenseID/releases/download/v0.13.0/rsid_api_docs.zip)

UnaNancyOwen, 2021-03-13:
@gabime That's great, however it is local file. Even better if it published on web page!

RealSenseSupport, 2021-03-22:
Thanks @UnaNancyOwen - Maybe we will be able to get to publishing on web but for now the zip will continue.

closed 2021-03-27 by RealSenseCustomerSupport