realsenseai/RealSenseID #17: RealSense ID SDK does not install by CMake

UnaNancyOwen, 2021-03-13:
Currentry, RealSense ID SDK does not install by CMake.
In general, the build files generated by CMake can generate INSTALL projects and install libraries. (cmake --build . --target install)
CMake files of RealSense ID SDK is missing scripts for this purpose.
(This issue is related to https://github.com/IntelRealSense/RealSenseID/issues/16)

RealSenseSupport, 2021-03-22:
Hi @UnaNancyOwen 

Is there anything further you need on this thread?

UnaNancyOwen, 2021-03-22:
@RealSenseSupport RealSense ID team working on related problem. It will be resolved with this issue shortly. 👍  https://github.com/IntelRealSense/RealSenseID/issues/16#issuecomment-803854474

RealSenseSupport, 2021-03-26:
Thanks for pointing this out...yes we are monitoring that one too.

gabime, 2021-04-01:
Released in v0.15.0. Use cmake -DRSID_INSTALL=ON to enable.

closed 2021-04-01 by gabime