realsenseai/RealSenseID #18: Very slow stdout

pjanco, 2021-03-13:
Hi,
I am working on porting this library for NXP IMXRT1062, NewlibNano and FreeRTOS environment. It is really not a easy task but and I am at good way to go.

Bu I have strange problem with very slow text outputs to stdout. I am using semihost debug console, I know semihost it is slow by design, but it is working fine at all my other projects .

1. If realsense library is not in project, then all text output to stdout are buffered. STD library is waiting for end of line to transmit whole buffer at once. This method is fast.
2. If realsense library is included in project, then this buffer is somehow disabled and every character is printed immediately and it is extremely slow. Problem is caused even if library is not used bud it is included and compiled in project.

I look manually at library sources but I did not found part of code that can possibly cause disabling of stdout buffer.

Can you guys know where is the problem?

Regards,
Peter.

gabime, 2021-03-13:
Maybe try disabling debug to console when building the lib:  ```cmake -DRSID_DEBUG_CONSOLE=OFF ..```

pjanco, 2021-03-13:
> Maybe try disabling debug to console when building the lib: `cmake -DRSID_DEBUG_CONSOLE=OFF ..`

I am not using cmake.
This symbol is undefined in my project settings. I  try to add new symbol `RSID_DEBUG_CONSOLE=OFF` to my project setting and it did not solve the problem.

I fund in realsense source code, this is used in some multiple parts of code: `#ifdef RSID_DEBUG_CONSOLE`.
So, I think it is correct to keep symbol undefined rather then defined it to OFF value.

gabime, 2021-03-13:
> I fund in realsense source code, this is used in some multiple parts of code: #ifdef RSID_DEBUG_CONSOLE.
So, I think it is correct to keep symbol undefined rather then defined it to OFF value.

Right, when not using CMake, the equivalent of  ```cmake -DRSID_DEBUG_CONSOLE=OFF``` is not to define RSID_DEBUG_CONSOLE

closed 2021-03-16 by gabime