realsenseai/RealSenseID #21: [feature request] Add Set Log Level

UnaNancyOwen, 2021-03-15:
RealSense ID SDK uses spdlog to output logs.
Currently, all levels of logs are output to console. It's annoying at times.
Please provide a way for user to dynamically control level of log output using environment variables.
spdlog provides such a feature. It is just adding <code>spdlog::cfg::load_env_levels()</code>.
https://github.com/gabime/spdlog/tree/v1.8.2#load-log-levels-from-env-variable-or-from-argv

What do you think? Thanks,

UnaNancyOwen, 2021-03-15:
Related https://github.com/IntelRealSense/RealSenseID/issues/18

gabime, 2021-03-15:
To control log level, the [SetLogCallback()](https://github.com/IntelRealSense/RealSenseID/blob/48ec75e88cb53aaaecd04f438bbc227aee619779/include/RealSenseID/Logging.h#L39) can be used (include "RealSenseID/Logging.h").
The following example displays only logs with level ```Info``` or higher:
```c++
#include "RealSenseID/Logging.h"
void my_log_clbk(RealSenseID::LogLevel level, const char* msg) 
{
    std::cout << msg << std::endl;    
}
// subscribe to log events with level Info or higher. 
RealSenseID::SetLogCallback(LogLevel::Info, true, my_log_clbk);
```

In addition, the following convenience CMake options can be used turn on/off debug to screen or to file:
* ```cmake -DRSID_DEBUG_CONSOLE=ON/OFF```: Debug all to screen. 
* ```cmake -DRSID_DEBUG_FILE=ON/OFF```: Debug all to to "rsid_debug.log" file.

UnaNancyOwen, 2021-03-15:
@gabime "log" that I pointed to here refers to the latter.
RealSense ID SDK that installed from installer will allways display debugging information on console, because it is compile with RSID_DEBUG_CONSOLE=ON. It can only be specified at compile time of SDK. It's not friendry for user.
My suggestion is to add <code>spdlog::cfg::load_env_levels()</code>. It will allow users to change their behavior at runtime without re-compile SDK by loading settings from environment variables.

gabime, 2021-03-15:
> If specify RealSenseID::LogLevel::Off to RealSenseID::SetLogCallback, but SDK will always display all log to console. Is it bug?

No, this is because the RSID_DEBUG_CONSOLE=ON 

> My suggestion is to add spdlog::cfg::load_env_levels(). It will allow users to change their behavior at runtime without re-compile SDK by loading settings from environment variables.

Good idea, will add it to our workplan.

UnaNancyOwen, 2021-03-15:
@gabime Thank you for discussion with me! 👍

gabime, 2021-07-26:
User code can implement any log filtering needed using the rsid logging API.

closed 2021-07-26 by gabime