realsenseai/RealSenseID #22: self-built rsid_csharp_example from source code does not display preview.

UnaNancyOwen, 2021-03-16:
The self-built rsid_csharp_example from source code does not display preview.
All other functions are working fine. But, preview is not display.
[image]

<code>_preview.Start(OnPreview);</code> returns False.
https://github.com/IntelRealSense/RealSenseID/blob/v0.13.0/examples/csharp_example/MainWindow.xaml.cs#L1159

Could you give me some advice? Thanks,

gabime, 2021-03-16:
To build with preview support, use the  ```-DRSID_PREVIEW=ON``` CMake option (and install opencv).

Please refer to the instructions in https://github.com/IntelRealSense/RealSenseID/tree/master/examples#how-to-build

UnaNancyOwen, 2021-03-16:
@gabime Thanks! I had missed that one. It works fine. :)

closed 2021-03-16 by UnaNancyOwen