realsenseai/RealSenseID #23: enrolled users will all remove when updating firmware

UnaNancyOwen, 2021-03-17:
Enrolled users will all remove when updating firmware.
Should we make backup of enrolled users data?
And on related, Is there a way to backup and restore enrolled users data?
Thanks,

jeyyoon, 2021-04-01:
@RealSenseSupport 
As I know, F455 takes only front face picture for enrollment when I got F455 first time.
But a few days later, I updated FW then I knew that it takes front, left and right face pictures for enrollment.

In this case, I think faceprints on database should be updated(deleted and re-enroll) as well.
Is it okay not to delete them and using them with new FW Version?

RealSenseSupport, 2021-04-05:
Hi @jeyyoon 

Anytime there is an update to the recognition model, then there is the requirement around re-enrolling users that have been enrolled already on the F455.  This is also required in the latest release as well 0.15.0 (https://github.com/IntelRealSense/RealSenseID/releases/tag/v0.15.0)

prasadnoolu, 2021-04-12:
can anybody tell me how to create a local database (in my pc ) to store the recognized face names with intel realsense faceID

RealSenseSupport, 2021-04-19:
There is a Host mode functionality that is provided in the SDK

You can look here although in the document you'll see server mode but we've renamed it as Host mode
https://github.com/IntelRealSense/RealSenseID#server-and-device-apis

In the getting started, you'll see in the Viewer settings page there is a feature called "HostMode" - https://www.intelrealsense.com/get-started-intel-realsense-id/
By setting this Host mode, when a face is enrolled it will be saved to a db on the local host system.

https://github.com/IntelRealSense/RealSenseID/blob/master/tools/rsid-viewer/Database.cs