realsenseai/RealSenseID #25: How can I use score in MatchFaceprints function.

donghyun999, 2021-03-22:
I want to use the score value used in the MatchFaceprints function.
Because in a database containing 20 faceprints, it's sometimes recognized as someone else. 
So I want to get the score and choose the person with the highest score out of 20 people.

UnaNancyOwen, 2021-03-22:
I think so too. Currently, there is no way to get the internally calculated score. It only display in the logs.
@gabime What do you think? Thanks,

alexk1976, 2021-03-22:
@donghyun999 it shouldn't happen. I want to debug and fix your issue. Maybe your faceprints from different FW versions? Can you re-enroll and reproduce using latest FW/app?
Regarding exposing score...we can do it but not sure it'll help you. We run huge benchmarks to calculate the score (threshold) to optimize FPR vs TPR values. I dont think you have ability to calculate required score. Am i wrong?

donghyun999, 2021-03-23:
We want to find the person with the highest score, can't we just use the score value?

alexk1976, 2021-03-23:
I see. we'll expose it in next version. If it's urgent ..you can do it yourself in MatchResult Matcher::MatchFaceprints() function. Add maxScore to MatchResult struct

donghyun999, 2021-03-24:
I think we can proceed with good development.
thank you for resposce

donghyun999, 2021-03-29:
i have a new question

donghyun999, 2021-03-29:
i have a new question

Face authentication succeeds if you only wear a mask, and face authentication succeeds when you wear a different hair style without wearing a mask. On the other hand, face authentication fails when wearing a mask and wearing a different hairstyle.

In this case, I think hairstyles also affect facial verification. I wonder why this happens.

[image]
[image]
[image]
[image]
[image]

alexk1976, 2021-03-29:
long hair might be in face area and captured along with other facial features. When only one part (mask or hair) changes ..the distance (match score) to original enrollment is still close and match is succeeded. When a few facial parts change the match score is not close enough and it fails. In next releases we'll introduce adaptive learning feature that will update original enrollment data and allow more facial feature changes on the way

closed 2021-04-06 by donghyun999