realsenseai/RealSenseID #26: It seems to take a long time to authenticate in host mode

donghyun999, 2021-03-23:
We use the MatchFacePrint function to authenticate 1000 people in host mode without [break](https://github.com/IntelRealSense/RealSenseID/blob/48ec75e88cb53aaaecd04f438bbc227aee619779/examples/cpp_example/main.cc#L424) . 
It takes about three to four seconds to automaticate once, but I wonder if it's normal to take this long or if there's a faster way.

alexk1976, 2021-03-23:
3-4 sec is too long time.. you supposed to get response within 1-2 seconds depends on the mode you are running. 
do you run in secure or non-secure mode? ExtractFaceprintsForAuthLoop API or regular ExtractFaceprintsForAuth?

donghyun999, 2021-03-24:
I used non-secure mode and ExtractFaceprintsForAuth function.

jeyyoon, 2021-03-30:
@alexk1976 
Hi. I've been looking into this issue and wonder how we can improve authentication speed in case many people are enrolled.

alexk1976, 2021-04-10:
we dont see such issue during our tests. what version you using? can you try latest version?

jeyyoon, 2021-04-12:
> we dont see such issue during our tests. what version you using? can you try latest version?

I used the latest version but without 'break' in Matcher::GetScores function.(with 1000 people vector)
Because we found that it's often wrong for the first person with a score over threshold.
If we don't break 'for' loop, it finds GT person.

jeyyoon, 2021-04-13:
@alexk1976 
Sorry I found that I measure time duration in a wrong way. Now I see it should not be like that.

closed 2021-04-16 by donghyun999