realsenseai/RealSenseID #27: maximum size of user id is magic number

UnaNancyOwen, 2021-03-23:
Currently, there is not defined constant, so the maximum size of a user id is magic number.
It is just only described in [parameter description](https://github.com/IntelRealSense/RealSenseID/blob/v0.13.0/include/RealSenseID/FaceAuthenticator.h#L98) of FaceAuthenticator::Enroll(). 
> Max user id size is 15 bytes (max total of 16 bytes including the terminating zero byte).

I think it should be defined as constant in RealSense ID SDK.
```cpp
namespace rsid
{
    constexpr int32_t MAX_USERID_LENGTH 16; // add define constant of this proposal
}
```
```cpp
// e.g. How to use constant in RealSenseID::FaceAuthenticator::QueryUserIds()
char** user_ids = new char*[number_of_users];
for (unsigned i = 0; i < number_of_users; i++){
    user_ids[i] = new char[rsid::MAX_USERID_LENGTH]; // use constant for pre-allocate output memory
}
authenticator->QueryUserIds( reinterpret_cast<char**>( user_ids.data() ), number_of_users );
```

---
BTW, I think this API is not very user-friendly. I think RealSenseID::FaceAuthenticator::QueryUserIds() should create an new API that takes <codde>std::vecotr<std::string>&</code> as an output argument. I think it is usefull allocate memory for output inside api. And, It will be automatically released, because it is vecotor container and string. It's give safe for user.
```cpp
std::vecotr<std::string> uder_ids;
authenticator->QueryUserIds( user_ids, number_of_users );
```
Of course, I know we can do something similar with current version. However, It is a little less than smart.
```cpp
std::vector<std::vector<uint8_t>> user_ids( number_of_users, std::vector<uint8_t>( rsid::MAX_USERID_LENGTH ) );
authenticator->QueryUserIds( reinterpret_cast<char**>( user_ids.data() ), number_of_users );
```

alexk1976, 2021-03-23:
Your request for constant for max length of user id  makes sense. Will add it to next release.
Regarding QuertUserIds() - Unfortunately MCU (embedded OSs) customers cannot use "std" types and asked us not to use any stl containers in API.

UnaNancyOwen, 2021-03-23:
> Regarding QuertUserIds() - Unfortunately MCU (embedded OSs) customers cannot use "std" types and asked us not to use any stl containers in API.

Very well. It makes sense. Thanks,

RealSenseSupport, 2021-04-09:
There was a change in the latest released https://github.com/IntelRealSense/RealSenseID/releases/tag/v0.15.0 regarding user id size.

closed 2021-04-27 by RealSenseCustomerSupport