realsenseai/RealSenseID #29: Compiling for ARM Linux / Support

mdisg, 2021-03-29:
Is RealSenseID SDK supported under ARM linux?

If so, how can I compile the library for an ARM based linux like an embedded system (e.g. buildroot/yocto)?

nickticket2u, 2021-03-29:
I did it on Raspberry Pi (ARM Linux)
just install cmake and g++ (or gcc  I can't recall).
go to src folder, type cmake ..
then type make -j
then it will generate librsid_c.so and librsid.so files
I was able to load the so file in my .net core linux arm project

mdisg, 2021-03-29:
@nickticket2u thanks but unfortunately that will not work for my case because my device does not have that much space to install build tools on it and buildroot does only provide the build tools for the host environment. I have to cross compile it using a toolchain released by the manufacturer.

nickticket2u, 2021-03-29:
I'm not well versed about all these arm linux thing. are these *.so files portable between arm linux systems ? the .so file itself is small

mdisg, 2021-03-29:
I think it depends on the libc/kernel version etc. on the device so it may not be compatible.

gabime, 2021-03-29:
> Is RealSenseID SDK supported under ARM linux?

No. However you can try to  compile/cross-compile it to your platform.

mdisg, 2021-03-29:
I managed to compile the examples without the preview feature.

If anyone else wants to cross compile I used a separate cmake file (toolchain-arm.cmake) containing the following variables:

```
SET(CMAKE_C_COMPILER   /path_to/arm-linux-gnueabi-gcc)
SET(CMAKE_CXX_COMPILER /path_to/bin/arm-linux-gnueabi-g++)
SET(CMAKE_FIND_ROOT_PATH  /path_to/target)
SET(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
SET(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
SET(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
```

Provide the toolchain to cmake with the following argument `-DCMAKE_TOOLCHAIN_FILE=toolchain-arm.cmake`

Could be useful for others if that information is added to the building instructions?

gabime, 2021-03-29:
@mdisg  That's great.  Could you provide more details on the platform it runs ?

mdisg, 2021-03-30:
@gabime sure it's an i.MX6 (ARM Cortex-A9) with 2 Cores and 512MB of RAM.
The linux system is build using buildroot with Kernel 4.9.88

I did some tests with the example c program. Sometimes there seems to be a serial communication issue e.g. reading the user ids from the device failes from time to time but if repeated right after a failure that it succeeds.  I will keep an eye on that.

Offtopic question is there a way to enroll/add users to the device with already recorded faceprints so that authentication can be done by the device? We want to enroll users on one of our devices with the real sense unit but spread it across many devices also having a real sense unit for authentication. I know there is host mode but I'm unsure about the performance if there are a lot of faceprints in the host database.

nickticket2u, 2021-03-30:
that's a good question, regarding the enrolling faceprints data

gabime, 2021-03-30:
> Offtopic question is there a way to enroll/add users to the device with already recorded faceprints

Currently there is not such feature in the library

alexk1976, 2021-03-30:
we have it in plan

mdisg, 2021-03-31:
@alexk1976 great! Can you give me a time range when it will likely be available?

Yesterday I tried to compile the SDK with the preview feature with opencv Version 4.3.0 which works for the librarys itself but not for the examples. It failes to link because it cannot find the opencv librarys. I could not figure it out why it does not find them.

I have added the opencv library path (separate folder) in `CMAKE_FIND_ROOT_PATH` and `CMAKE_LIBRARY_PATH` but both did not solve it.

KennyDizi, 2021-04-22:
looking forward the SDK support ARM Linux (run on AIoT motherboard)

RealSenseSupport, 2021-05-15:
Hi @mdisg 

Have you tried the latest release? Additional changes to support cross platform compatibility have been introduced since.

Thanks

mdisg, 2021-05-17:
@RealSenseSupport There is nothing mentioned in the release notes. What exactly has change for cross platform compatibility? Do you mean the removal of opencv? If that's the case that was a good choice. Compiling the latest `0.17.0` SDK was possible using my existing toolchain cmake file. No errors in compiling the samples.

gabime, 2021-07-26:
>  What exactly has change for cross platform compatibility?
 
Indeed OpeNCV dependency was removed, and also removed the dependency on mutexes so it can now compiled in environments without proper thread support.

closed 2021-07-26 by gabime