realsenseai/RealSenseID #30: sometimes preview image is very dark

UnaNancyOwen, 2021-03-31:
Sometimes preview image is very dark.
Once it goes dark, it will stay dark until the next time authentication. (related IR flash light?)
It seems automatic exposure does not work well.
Can anyone reproduce this problem? and, Is it possible to fix this problem by updating firmware?
Thanks,

UnaNancyOwen, 2021-03-31:
related https://github.com/IntelRealSense/RealSenseID/issues/12

alexk1976, 2021-03-31:
You are right. latest drop not using auto-exposure for preview. Auto-exposure drop is expected on this Thursday

RealSenseSupport, 2021-04-05:
Hi @UnaNancyOwen 

Have you tried the 0.15.0 SDK release https://github.com/IntelRealSense/RealSenseID/releases/tag/v0.15.0 that now has auto exposure enabled?

UnaNancyOwen, 2021-04-06:
Yes, It seems to be works fine! Thanks,

closed 2021-04-06 by UnaNancyOwen