realsenseai/RealSenseID #33: request changelog

UnaNancyOwen, 2021-04-02:
[v0.15.0](https://github.com/IntelRealSense/RealSenseID/releases/tag/v0.15.0) has been released.
Are there any details about the changes from v0.13.0 to v0.15.0?

I found a few big changes. These are important pieces of information for developers.

* removed serialPortType from SerialConfig (it is destructive change!)
* add FHD_Rect to PreviewMode
* add advanced_mode to DeviceConfig
* add FaceRect to Image

Please release changelog! I'm afraid undocumented changes.

---
BTW, [this](https://github.com/IntelRealSense/RealSenseID/pull/31) is too many changes in one commit. It is hard to track changes for developer.

alexk1976, 2021-04-02:
Hi,
Most of the changes published in release notes.
Why do you need portType in addition to port number? Our library working seamless if it's USB or UART. Moreover we have auto-detection of port type.
The rest features you mentioned (FHD_Rect, advanced_mode...) they are not documented by purpose as available only to specific customers with special license.  Published FW doesnt support them.
RecognitionOnly mode not running Anti-Spoofing algo(s) as part of authentication. 
Auto-exposure is controlled automatically by FW and starts upon preview. Users dont have control for it.
Currently all development is done in an internal github, then fully validated and the release is published to the public github.
In future we plan to establish automated CI and push smaller commits directly to public github

UnaNancyOwen, 2021-04-02:
> Why do you need portType in addition to port number? Our library working seamless if it's USB or UART. Moreover we have auto-detection of port type.

No, I'm not saying that change is bad. I think that change is a good one.
However, I'm afraid **undocumented** changes. That change is not even listed in the release notes.
We didn't know about that change until we read the code.

> The rest features you mentioned (FHD_Rect, advanced_mode...) they are not documented by purpose as available only to specific customers with special license. Published FW doesnt support them.

Haha, Yes, that's it! I want a document with such an explanation.
We didn't even know that the feature was not available to the public until your comment.

> RecognitionOnly mode not running Anti-Spoofing algo(s) as part of authentication.
> Auto-exposure is controlled automatically by FW and starts upon preview. Users dont have control for it.

Good! It would be better if this is documented in changelog.

closed 2021-07-26 by gabime