realsenseai/RealSenseID #38: How can I get wearing mask status and the location of the face?

donghyun999, 2021-04-12:
We do face verification with F455.

it is influenced a lot by wearing a mask.

If you wear a mask, other people's scores can be recognized as a threshold.

So we wonder if there is a way to know if a person is wearing a mask and where the face is in the picture.

RealSenseSupport, 2021-04-19:
Hi @donghyun999 

Can you provide image of the scenario that you are outlining?  That way we have a better understanding of your question and/or feature request.

donghyun999, 2021-04-20:
[image] <wearing mask>
[image] <none wearing mask>
if you're wearing a mask or not.

closed 2021-04-26 by donghyun999