realsenseai/RealSenseID #44: RealSenseID::FaceAuthenticator::Cancel() ,integration , segfaults

Baylibrejmonnet, 2021-04-22:
### plateform
* `Linux xxx 5.8.0-50-generic #56~20.04.1-Ubuntu SMP Mon Apr 12 21:46:35 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`
* Ubuntu 20.04
* VSCODE. 1.55
* Clang 10.0.0 x86_64-pc-linux-gnu

hi , i'm new with you device.
i'm having fun with the the client (rsid-cli).
at this point i'm having trouble using `uto status = authenticator->AuthenticateLoop(auth_clbk);` and `authenticator->Cancel();`

i changed the `do_anthenticate` function from _main.c_ .

=>>>
`void do_authenticate(const RealSenseID::SerialConfig& serial_config)
{
    system("paplay  /usr/share/sounds/freedesktop/stereo/power-plug.oga ");
    auto authenticator = CreateAuthenticator(serial_config);
    MyAuthClbk auth_clbk;
    auto status = authenticator->AuthenticateLoop(auth_clbk);
    authenticator->Cancel();
    
    if (status != RealSenseID::Status::Ok)
    {
        
        std::cout << "Status: " << status << std::endl << std::endl;
    }
    else {
        system("paplay  /usr/share/sounds/freedesktop/stereo/bell.oga");
    }
}`

at this point , if read correctly the `cancel()`, should be called when i have the status Ok.
but nothing happend i stay stuck in the loop until i've the famous segfault .
do i set the `cancel() `method correctly or is something else ? 

`[2021-04-23 00:34:36.513] [debug] [PacketSender] Waiting packet..
[2021-04-23 00:34:38.954] [debug] [PacketSender] Received packet 'g' after 2441 millis
[2021-04-23 00:34:38.955] [debug] [FaceAuthenticatorImpl] Detected face 508,610 466x518
[2021-04-23 00:34:38.955] [info] [DetectSpoof] OnFaceDetected 1 faces
[2021-04-23 00:34:38.955] [debug] [PacketSender] Waiting packet..
[2021-04-23 00:34:39.556] [debug] [PacketSender] Received packet 'R' after 600 millis
[2021-04-23 00:34:39.556] [info] [Autenticate] OnResult status=Success(0), user_id="john"
******* Authenticate success.  user_id: john *******
[2021-04-23 00:34:39.556] [debug] [PacketSender] Waiting packet..
[2021-04-23 00:34:40.450] [debug] [PacketSender] Received packet 'g' after 894 millis
[2021-04-23 00:34:40.450] [debug] [FaceAuthenticatorImpl] Detected face 522,642 406x496
[2021-04-23 00:34:40.451] [info] [DetectSpoof] OnFaceDetected 1 faces
[2021-04-23 00:34:40.451] [debug] [PacketSender] Waiting packet..
[2021-04-23 00:34:40.697] [debug] [PacketSender] Received packet 'H' after 246 millis
[2021-04-23 00:34:40.697] [info] [Autenticate] OnHint status=NoFaceDetected(1), user_id=""
on_hint: hint: NoFaceDetected
[2021-04-23 00:34:40.697] [debug] [PacketSender] Waiting packet..
[2021-04-23 00:34:43.883] [debug] [PacketSender] Received packet 'H' after 3185 millis
[2021-04-23 00:34:43.883] [info] [Autenticate] OnHint status=DeviceError(15), user_id=""
on_hint: hint: DeviceError
[2021-04-23 00:34:43.883] [debug] [PacketSender] Waiting packet..
[2021-04-23 00:34:43.883] [debug] [PacketSender] Received packet 'R' after 0 millis
[2021-04-23 00:34:43.883] [info] [Autenticate] OnResult status=Forbidden(14), user_id=""
on_result: status: Forbidden
[2021-04-23 00:34:43.883] [debug] [PacketSender] Waiting packet..
[2021-04-23 00:34:43.892] [debug] [PacketSender] Received packet 'Y' after 8 millis
[2021-04-23 00:34:43.892] [info] [Autenticate] Done
[2021-04-23 00:34:43.892] [debug] [NonSecureSession] Cancel requested.
[2021-04-23 00:34:44.146] [debug] [NonSecureSession] Close session
[2021-04-23 00:34:44.146] [debug] [NonSecureSession] Sending cancel..
Erreur de segmentation (core dumped)`
[rsid_debug.log](https://github.com/IntelRealSense/RealSenseID/files/6361622/rsid_debug.log)

gabime, 2021-04-23:
> at this point , if read correctly the cancel(), should be called when i have the status Ok.
but nothing happend i stay stuck in the loop until i've the famous segfault .
do i set the cancel() method correctly or is something else ?

The `Cance()` should  be called during the session (from one of the callback functions or from a different thread), not after the function returns.

As for the segfault, fixed in c3cf9e4856992a4a3a5bee8a0c68918eeb3d7eb6. Please take latest commit and see if it solves your problem.

Baylibrejmonnet, 2021-04-23:
thank for the fix, @gabime 
i do not have anay segfault now .
and now `cancel()` is working thank to you observation .

```
void do_authenticate(const RealSenseID::SerialConfig& serial_config)
{
    system("paplay  /usr/share/sounds/freedesktop/stereo/power-plug.oga ");
    auto authenticator = CreateAuthenticator(serial_config);
    MyAuthClbk auth_clbk(authenticator.get());
    authenticator->AuthenticateLoop(auth_clbk);

}
```

```
// Authentication callback
class MyAuthClbk : public RealSenseID::AuthenticationCallback
{
    RealSenseID::FaceAuthenticator* _authenticator;
public:
    MyAuthClbk(RealSenseID::FaceAuthenticator* authenticator): _authenticator(authenticator)
    {
    }
    void OnResult(const RealSenseID::AuthenticateStatus status, const char* user_id) override
    {
        if (status == RealSenseID::AuthenticateStatus::Success)
        {
            std::cout << "******* Authenticate success.  user_id: " << user_id << " ****CANCEL***" << std::endl;
            _authenticator->Cancel();
        }
        else
        {
            system("paplay /usr/share/sounds/freedesktop/stereo/dialog-error.oga ");
            std::cout << "on_result: status: " << status << std::endl;
        }
    }
```

gabime, 2021-04-23:
Great. Thanks for reporting.

closed 2021-04-23 by gabime