realsenseai/RealSenseID #46: Is it okay to use multiple faceprints average value from one person?

jeyyoon, 2021-04-27:
We keep testing f455 performance and we found it recognize a person as another person.
We assume that the moment when f455 take a picture is not the best time.

So F455 takes multiple pictures and I wonder it is okay to use average value of multiple faceprints(element-wisely)
When I see the code of calculating similarity(f/w 0.16 version)
I guess it probably okay but want to know F455 developers opinion.

Thank you

alexk1976, 2021-05-01:
this issue should not happen (match a wrong person). please provide more details . is it device or host mode? are these persons (enrolled and "false" authenticated) indeed look very similar?  do you use our application or your own?

jeyyoon, 2021-05-02:
@alexk1976
We used both the demo program provided by Intel and SDK. of course, no code change for authentication(match and similarity calculation)
The situation happened both device and host mode. We think two people are not similar each other but their hair style was similar at that time.(one of them had his hair cut and it rarely happen again)

Especially, in case people wear a mask, it's confused more.
For us, face recognition system is really important so I'm trying to find any better method to make it perform better.

Could you please tell me your opinion to use above idea?(using average faceprints)

alexk1976, 2021-05-02:
Actually we going to release such feature (adaptive learning based on average)  today or tomorrow. please reproduce the issue and update us. If issue is reproduced ..i suggest joint debugging with us as such issue should not happen. We'll enable dump of images and debug together

jeyyoon, 2021-05-03:
@alexk1976 
Thanks. I could see a new commit yesterday. I wish it's getting better.
We are going to use F455 at our customer fields and hundreds people will use our program with RSID at each field.

Especially some of our customers are students at schools and we know that if they wear a mask, many of them look similar. So we consider to ask them to wear off mask(or move down below nose) at this stage.

We hope Intel can provide API function that tell us the person is wearing a mask in front of F455 and it would be much better if it tells the person needs to move down his/her mask.

jeyyoon, 2021-05-03:
To analyze when it recognizes people well or when it doesn't. 
Our program saves people face images during the enrollment & authentication.
But we have no choice but only before or after calling API. 
Can we receive the real RGB array which is used for enrollment and authentication?

alexk1976, 2021-05-06:
Please provide contact and use-case high-level details to alex.kunin@intel.com, nick.nam@intel.com and james.l.scaife@intel.com emails.
We may provide you special version enabling dumps and debugging

closed 2021-07-23 by alexk1976