realsenseai/RealSenseID #47: Firmware ERRO: What do I do?

cabelo, 2021-04-28:
The new release does not work without updating the firmware. But when I update this error of the images occurs.

[image]

[image]

[image]

[image]

alexk1976, 2021-05-06:
Please use latest SDK and upgrade to bundled FW
Let me know if the issue still exists

cabelo, 2021-05-12:
Hi @alexk1976 Problem continues, below the evidence.
[image]
[image]
[image]

benlev, 2021-05-12:
Hi @cabelo, it looks like you paired your device after compiling with RSID_SECURE=1. This is done automatically when running the rsid-viewer tool, or manually, if you use the rsid-cli tool.

If you used the sample set of keys, you can simply re-compile with RSID_SECURE=1 and try again. However, if you paired with your own set of keys, you must use them.

To be able to work with the unit with RSID_SECURE=0, you must first unpair the device. To do so, run the rsid-cli sample and select 'i'. Again, this should be done with the set of keys you used when pairing.

Let me know if this works out for you.

-Ben

cabelo, 2021-05-13:
Ok  @benlev 

I'm going to do this, do you prefer this test of RSID_SECURE = 0 in linux or windows?

But I would like to clarify, that I used only this installer program : rsid_sdk_v0.17.1_win64.exe

cabelo, 2021-05-13:
Hi @benlev 

I build  RSID_SECURE=0 on the linux, bellow see the erros

[image]

[image]

benlev, 2021-05-13:
@cabelo, please try the following:
1. Compile with RSID_SECURE=1.
2. Run 'rsid-cil' and select 'p' to pair with the device.
3. Try enrollment/authentication - should now work.
4. Select 'i' to unpair the device.
5. Re-compile with RSID_SECURE=0.
6. Run 'rsid-cli' and try enrollment/authentication.

-Ben

cabelo, 2021-05-17:
@benlev 

Initially I tested it in Realsense ID Viewer to test the functionality and do not run the risk of having problems.

I recompiled with the parameter RSID_SECURE = 1 and everything worked on linux, but in the Realsense ID Viewer program on Windows the problem continues.
I will particularly work only on Linux, but I think it would be interesting for everything to work on both platforms.

If I had started on Linux, I wouldn't have even opened this request.
I will particularly work only on Linux, but I think it would be interesting for everything to work on both platforms.

If I had started on Linux, I wouldn't have even opened this request.

Below the evidence, thank you.

First In Linux:

[image]

[image]

[image]

But... In Windows...

[image]

benlev, 2021-05-18:
@cabelo, when running rsid-viewer on Windows, are you using the SDK installer or compiling from source?

cabelo, 2021-05-18:
> @cabelo, when running rsid-viewer on Windows, are you using the SDK installer or compiling from source?

Hi @benlev, i used  the SDK installer

cabelo, 2021-05-19:
> > @cabelo, when running rsid-viewer on Windows, are you using the SDK installer or compiling from source?
> 
> Hi @benlev, i used the SDK installer

@benlev Other important information, with the parameter RSID_SECURE = 0 enroll and authenticate does not work.

benlev, 2021-05-19:
@cabelo 
The SDK installer is compiled with RSID_SECURE=0, which means your device must be unpaired to be able to communicate with it.

Since you used the sample keys to pair with your device, you have 2 options:
1. Compile the solution on Windows using RSID_SECURE=1 and run 'rsid-viewer'.
2. Run 'rsid-cli' in your Linux environment, hit 'i' to unpair and 'q' to exit the application. Connect the device to your Windows environment and run the 'rsid-viewer' that was installed with the SDK installer.

cabelo, 2021-05-20:
@benlev  Thanks, all work now.

I take this opportunity to show that as an openSUSE ambassador and founder of the openSUSE Innovators initiative, I homologated the  RealSenseID on all openSUSE Linux.

See: 
https://news.opensuse.org/ 
https://news.opensuse.org/2021/05/19/RealSenseID-available-for-openSUSE/
https://en.opensuse.org/SDB:Install_RealSenseID

closed 2021-06-16 by RealSenseCustomerSupport