realsenseai/RealSenseID #49: F455. Device failed to respond. Error message.

PavKovalenko, 2021-05-03:
Try to connect newly bought camera IntelRealSense F455 to Android and Windows, but failed.

When connectto Andoid I see in logs:
```
I/librs Enumerator: onReceive: android.hardware.usb.action.USB_DEVICE_ATTACHED
W/librs UsbUtilities: getUsbDevice: failed to locate USB device, VID: 0x8086, PID: 0x0000
I/librs MessagesHandler: handleMessage: realsense device attached
I/librs Enumerator: notifyOnAttach
```
When I launch RS camera app from PlayStore. It cannot see device as well.

After that tried to connect camera to Windows. And launch last Intel RealsSense viewer for Mac. And got this message.
[image]

What I am doing wrong. Please help

alexk1976, 2021-05-06:
1. For F455 you must use RealSenseID SDK/releases only from this github and not libRS
2. Start from Windows setup, download latest SKD and upgrade to latest FW bundled inside of SDK
let us know if you still have an issue

PavKovalenko, 2021-05-07:
I used last version of SDK from
https://github.com/IntelRealSense/RealSenseID/releases/tag/v0.17.1
Same screen. How I can upgrade FW to latest?

When I launch on Android I see
[image]

But no any preview

PavKovalenko, 2021-05-07:
I updated camera firmware to latest 0.17 and it helps. Currently it works on Windows and Android as well
Thank you a lot