realsenseai/RealSenseID #50: I think I bricked my device!

cedivad, 2021-05-04:
There is no sugar coating it, I might be a dumbass. I believe I paired the device with a made up public key. Here is what I did:

```
-----BEGIN EC PARAMETERS-----
BgUrgQQACg==
-----END EC PARAMETERS-----
-----BEGIN EC PRIVATE KEY-----
MHQCAQEEIPOH4AS6mqV4E7uidBBToCLTe/kNCVqu4GSCWcNqyZZLoAcGBSuBBAAK
oUQDQgAEotfPBi0ilctH4+l/rWC2uJm4TYDIYo9r0l69VMPGsM8ltOfUVrDgO4N0
7Zcf01/tezS6J6TnE+DotgcqNLlqbQ==
-----END EC PRIVATE KEY-----
```

Commands to extract the hex priv/pub keys from it:

```
openssl ec -in key.pem -outform DER|tail -c +8|head -c 32|xxd -p -c 32
openssl ec -in key.pem -pubout -outform DER|tail -c 64|xxd -p -c 64
```

I then compiled with:

```
// Example of host's private key. Should be replaced in production with your own secret key
static unsigned char SAMPLE_HOST_PRI_KEY[PRI_KEY_SIZE] = {0xf3, 0x87, 0xe0, 0x04, 0xba, 0x9a, 0xa5, 0x78, 0x13, 0xbb, 0xa2, 0x74, 0x10, 0x53, 0xa0, 0x22, 0xd3, 0x7b, 0xf9, 0x0d, 0x09, 0x5a, 0xae, 0xe0, 0x64, 0x82, 0x59, 0xc3, 0x6a, 0xc9, 0x96, 0x4b};

// Example of host's public key. Should be replaced in production with your own key
static unsigned char SAMPLE_HOST_PUB_KEY[PUB_KEY_SIZE] = {
    0xa2, 0xd7, 0xcf, 0x06, 0x2d, 0x22, 0x95, 0xcb, 0x47, 0xe3, 0xe9, 0x7f, 0xad, 0x60, 0xb6,
    0xb8, 0x99, 0xb8, 0x4d, 0x80, 0xc8, 0x62, 0x8f, 0x6b, 0xd2, 0x5e, 0xbd, 0x54, 0xc3, 0xc6, 0xb0,
    0xcf, 0x25, 0xb4, 0xe7, 0xd4, 0x56, 0xb0, 0xe0, 0x3b, 0x83, 0x74, 0xed, 0x97, 0x1f, 0xd3, 0x5f,
    0xed, 0x7b, 0x34, 0xba, 0x27, 0xa4, 0xe7, 0x13, 0xe0, 0xe8, 0xb6, 0x07, 0x2a, 0x34, 0xb9, 0x6a, 0x6d};
```

Pairing succeeded, but I could never unpair.

Looking at the code it seems it's signing my public key using the device's pub key and then sending it, but there there is no checking the key I inserted is valid.

How can I make sure I don't repeat my mistake again? Is there a documented procedure somewhere to extract the hex from a pem file?

Also is there a secure enclave chip I can replace on the board?

Many thanks,
Davide

cedivad, 2021-05-07:
I tried retracing my steps, but couldn't find any mistake. Gonna have to try again with the libraries this is shipping with.

cedivad, 2021-05-07:
I could use the code above to re-generate the sample pubkey the library ships with after changing the curve to NIST256p, which verified this isn't an endianness problem, so I really, really don't understand what the problem could be. (I can't see how a different curve might be it).

At least now I know that my device is paired to a pubkey I have the privkey of.

```
from ecdsa import SigningKey, SECP256k1, NIST256p
import binascii, ecdsa

#private_key = "f387e004ba9aa57813bba2741053a022d37bf90d095aaee0648259c36ac9964b"
private_key = "b9adfe0e6dd4fb6f76df5392874e5839dd51d1aa79945ea8368fb5dfa8282653"

private_key = bytes(private_key, 'utf-8')
private_key = binascii.unhexlify(private_key)
priv = SigningKey.from_string(private_key, curve=NIST256p)
pub = priv.get_verifying_key().to_string()

print(pub.hex())
```

cedivad, 2021-05-08:
I tried changing some of the SECP256R1 references in the sdk to K1, and it's still not working, I'm assuming it's because the device's firmware is expected to use R1 curves only. I guess it is a curves problem indeed then. 

I wish I read about it somewhere before bricking it!

benlev, 2021-05-12:
Hi @cedivad,
Just to make sure I understand - the key that the device is currently paired with was generated using a K1 curve?

cedivad, 2021-05-12:
Yes, that is correct. Apologies for the messy wall of text above.

RealSenseSupport, 2021-05-25:
Hi @cedivad 

When you purchased the camera, did you purchase through the intelrealsense.com webstore or did you purchase through distributor?  Please submit an RMA request through one of those channels.

closed 2021-06-18 by RealSenseCustomerSupport