realsenseai/RealSenseID #51: Does adaptive learning make system better?

jeyyoon, 2021-05-06:
At the newest feature 0.17.1, I found a new feature of
<Adaptive learning. Update the user's faceprints when new information can be learned, after a successful authentication attempt.>

It seems like it updates a user's faceprint after the user does authentication process.
But, what if F455 does wrong?(mean it recognizes the person as another one)
If it happens it makes itself worse I guess. 
Or, could I get some explanation this concept?

alexk1976, 2021-05-06:
we have a few methods to avoid it:
1. we dont update faceprints in case we are not absolutely sure it's same user using high threshold
2. if after update faceprints match score is too far from original enrollment we'll update it to original enrollment

jeyyoon, 2021-05-06:
Thanks for letting us know how to avoid the problem I'm worried about.

I designed our faceprints database before, like one person can enroll two faceprints.
One is from enrollment process and the other is from authentication process.

It is obviously not a good way though, after authentication process, we ask to people like 'Are you James, or who are you?'.
And we get the answer use update the person's faceprint in the answer.
So database keeps people's enrollment faceprints and updates people's faceprints from authentication process.

But if Intel RSID team can say if there is NICE threshold value to be sure who the person is, we can apply that concept into our application.

Or could you give us any advice?
Thanks

alexk1976, 2021-07-23:
please update if issue was solved for you

closed 2021-07-26 by gabime