realsenseai/RealSenseID #52: Stuck in Authentication Loop and Viewer Application Crash

mdisg, 2021-05-06:
I have updated to the latest SDK Version `0.17.1` and Firmware `F450_2.6.0.74_SIGNED.bin`.

During testing with the new SDK I started the authentication using the method `AuthenticateLoop`.
After that I aborted the debugging session within Visual Studio without calling Cancel or something else.
The authentication mode is then stuck, the red LED does continue to flash. I then tried to start the sample application `rsid-viewer.exe` which immediately crashes without an error message on start. The program does not show up. I have to unplugg the device to get it out of that authentication loop.

It seems so that calling `Cancel `(which I did in my program) does not have any effect.

alexk1976, 2021-05-06:
You need to invoke Cancel in the same session. In next releases we plan to handle the case when app/user doesnt do it and we get stuck

mdisg, 2021-05-06:
@alexk1976 I'm aware of that but there are cases in which you cannot guarantee that `Cancel` is called like a crash of the application like `OutOfMemoryException` or something like that. But nice to here that you have it in mind and the next release does handle it. Thanks for that information. 

Offtopic: Compilation for linux arm platform does work better with the new SDK as with the old one I tried a while ago. Removal of opencv was a good choice!

alexk1976, 2021-07-23:
Issue was solved in previous releases

closed 2021-07-23 by alexk1976