realsenseai/RealSenseID #54: [PacketSender] fail to get the right recv message

lmjieSCU, 2021-05-12:
I run the sample file : samples/cpp/enroll.cc in ubuntu20.10 directly. And  when the PacketSender send the head packet,I get only 3 bytes [400x 460x  6f0x]/[@ f o](%u).but here should get 19 bytes and the protocol_ver should be 20x but not 6f0x.so I got the versionnotmatch error.
help!TKS!

RealSenseSupport, 2021-05-18:
Could you double check the FW version in your device?
Could you provide more detail about the environment?
You have to use the RSID with the correct FW.

lmjieSCU, 2021-05-19:
yeah, get it

lmjieSCU, 2021-05-19:
I update the FW version to F450_2.6.0.74 through the rsid-fw-update  successfully,but when run the rsid-cli,I got something wrong.
the report is:
[2021-05-19 03:31:51.129] [debug] [LinuxSerial] Opening serial port /dev/ttyACM0 baudrate 115200
Connected to device
[2021-05-19 03:31:51.208] [debug] [NonSecureSession] Start session
[2021-05-19 03:31:51.210] [debug] [PacketSender] Sending packet 'o'
[2021-05-19 03:31:51.215] [debug] [PacketSender] Waiting packet..
[2021-05-19 03:31:51.231] [debug] [PacketSender] Received packet 'Y' after 15 millis
[2021-05-19 03:31:51.231] [error] [NonSecureSession] Failed to recv device start session response
[2021-05-19 03:31:51.232] [debug] [PacketSender] Sending packet 'q'
[2021-05-19 03:31:51.238] [debug] [PacketSender] Waiting packet..
[2021-05-19 03:31:51.254] [debug] [PacketSender] Received packet 'Y' after 15 millis
[2021-05-19 03:31:51.254] [error] [NonSecureSession] Invalid sequence number. Last: 0, Current: 0
[2021-05-19 03:31:51.254] [error] [FaceAuthenticatorImpl] Failed receiving fa packet (status 106)
Status: SerialSecurityError

the version information is :
Additional information:
 * S/N: 04946228053799A27FM10464-100
 * Firmware: DNET:4.5.20.0|YOLO:25.4.21.0|AS2DLR:23.2.21.0|RECOG:1.2.21.0|OPFW:2.6.0.74|NNLED:2.22.21.0|NNLAS:2.12.21.0
 * Host: 0.17.1

lmjieSCU, 2021-05-19:
I notice the RECOG version is 1.2.21.0 not 28.2.21.0

lmjieSCU, 2021-05-19:
whatever packet I send ,the recv is always begin with Y

RealSenseSupport, 2021-05-19:
This is the info in my device.

Additional information:
 * S/N: xxxxxxxxxxxxxxxxxxxxxxxx-100
 * Firmware: DNET:4.5.20.0|YOLO:25.4.21.0|RECOG:28.2.21.0|AS2DLR:23.2.21.0|OPFW:2.6.0.74|NNLED:2.22.21.0|NNLAS:2.12.21.0
 * Host: 0.17.1

I think you have bad FW in your device.
I use the bin file from the installer package. which file do you use?

lmjieSCU, 2021-05-19:
ok...I change a new device and update it in windows.It work successly

RealSenseSupport, 2021-06-04:
Thanks for your feedback..

closed 2021-06-04 by RealSenseCustomerSupport