realsenseai/RealSenseID #66: CMake configuration with --DRSID_PREVIEW=1 fails

michalrudowicz, 2021-05-21:
Issue happens both when building on Fedora 33 and in Ubuntu 20.04 (in Docker environment).

I believe it may at least partially be caused by the lack of the `cmakescripts` directory in `libjpeg-turbo_2_1_0` directory. [This directory exists in the upstream libjpeg-turbo repository](https://github.com/libjpeg-turbo/libjpeg-turbo/tree/2.1.0/cmakescripts).

Problem doesn't happen when this directory is copied verbatim from the upstream repository.

As a side note, maybe git submodules would be easier to deal with?

Build log:
```
❯ cmake .. -DRSID_PREVIEW=1 
-- CCache found!
-- Build type: Release
-- Linux platform
-- CMAKE_BUILD_TYPE = Release
-- VERSION = 2.1.0, BUILD = 20210521
-- 64-bit build (x86_64)
-- CMAKE_INSTALL_PREFIX = /opt/RealSenseID
CMake Error at 3rdparty/libjpeg-turbo_2_1_0/CMakeLists.txt:135 (include):
  include could not find load file:

    cmakescripts/GNUInstallDirs.cmake

-- CMAKE_INSTALL_BINDIR =  ()
-- CMAKE_INSTALL_DATAROOTDIR =  ()
-- CMAKE_INSTALL_DOCDIR =  ()
-- CMAKE_INSTALL_INCLUDEDIR =  ()
-- CMAKE_INSTALL_LIBDIR =  ()
-- CMAKE_INSTALL_MANDIR =  ()
-- Shared libraries disabled (ENABLE_SHARED = 0)
-- Static libraries enabled (ENABLE_STATIC = 1)
-- 12-bit JPEG support disabled (WITH_12BIT = 0)
-- Arithmetic decoding support enabled (WITH_ARITH_DEC = 1)
-- Arithmetic encoding support enabled (WITH_ARITH_ENC = 1)
-- TurboJPEG API library disabled (WITH_TURBOJPEG = 0)
-- TurboJPEG Java wrapper disabled (WITH_JAVA = 0)
-- In-memory source/destination managers enabled (WITH_MEM_SRCDST = 1)
-- Emulating libjpeg API/ABI v6.2 (WITH_JPEG7 = 0, WITH_JPEG8 = 0)
-- libjpeg API shared library version = 62.3.0
-- Compiler flags =  -O3 -DNDEBUG
-- Linker flags =  
-- Compiler supports pointers to undefined structures.
-- INLINE = __inline__ __attribute__((always_inline)) (FORCE_INLINE = 1)
-- Linker supports GNU-style version scripts
-- CMAKE_EXECUTABLE_SUFFIX = 
-- SIMD extensions: None (WITH_SIMD = 0)
-- FLOATTEST = no-fp-contract
CMake Error at 3rdparty/libjpeg-turbo_2_1_0/CMakeLists.txt:1473 (install):
  install PROGRAMS given no DESTINATION!

CMake Error at 3rdparty/libjpeg-turbo_2_1_0/CMakeLists.txt:1475 (install):
  install PROGRAMS given no DESTINATION!

CMake Error at 3rdparty/libjpeg-turbo_2_1_0/CMakeLists.txt:1477 (install):
  install PROGRAMS given no DESTINATION!

CMake Error at 3rdparty/libjpeg-turbo_2_1_0/CMakeLists.txt:1484 (install):
  install FILES given no DESTINATION!

CMake Error at 3rdparty/libjpeg-turbo_2_1_0/CMakeLists.txt:1514 (install):
  install FILES given no DESTINATION!

CMake Error at 3rdparty/libjpeg-turbo_2_1_0/CMakeLists.txt:1519 (include):
  include could not find load file:

    cmakescripts/BuildPackages.cmake

CMake Error: File /path/to/RealSenseID/3rdparty/libjpeg-turbo_2_1_0/cmakescripts/cmake_uninstall.cmake.in does not exist.
CMake Error at 3rdparty/libjpeg-turbo_2_1_0/CMakeLists.txt:1521 (configure_file):
  configure_file Problem configuring file

-- Configuring incomplete, errors occurred!
See also "/path/to/RealSenseID/b2/CMakeFiles/CMakeOutput.log".
See also "/path/to/RealSenseID/b2/CMakeFiles/CMakeError.log".
```

gabime, 2021-05-21:
Fixed. Thanks @michalrudowicz for reporting.

closed 2021-05-21 by gabime

michalrudowicz, 2021-05-21:
Sorry for the confusion, but it looks like the `release` directory ([upstream](https://github.com/libjpeg-turbo/libjpeg-turbo/tree/2.1.0/release)) is missing in the same way.

gabime, 2021-05-21:
Fixed. Thanks for spotting this.