realsenseai/RealSenseID #72: Crash when running rsid-viewer.exe

wolfysan, 2021-05-25:
OS: Window 10
Camera: F455
CPU: AMD
-----------------------------------
I just downloaded the latest version v0.18.1, I installed it on my Win 10 PC, and when I run the Intel Realsense Viewer, the window appears, and after a few moments, it closes. 

I checked EventViewer and I saw an error there without to many details:
Application: rsid-viewer.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.AccessViolationException
   at rsid.Authenticator.rsid_is_fw_compatible_with_host(System.String)
   at rsid_wrapper_csharp.MainWindow.QueryDeviceMetadata(rsid.SerialConfig)
   at rsid_wrapper_csharp.MainWindow.DetectDevice()
   at rsid_wrapper_csharp.MainWindow.InitialSession(System.Object)

What could be the cause of this problem?

Thank you.

3dsf, 2021-05-25:
In my limited experience with this product, I found it is very important to match the firmware version to the sdk version.  
The fw can be found in the assets section, https://github.com/IntelRealSense/RealSenseID/releases

gabime, 2021-05-25:
There was a recent pr #69 that might be fix this issue as well. Please try to recompile the viewer using latest commit and see if it solves the issue.

wolfysan, 2021-05-27:
He found that the firmware is old, and we are currently working on updating it.
Depending on what happens after the firmware update, I will return an update this issue.

wolfysan, 2021-05-31:
Update firmware fixed the issue.

closed 2021-05-31 by wolfysan