realsenseai/RealSenseID #76: F455 - Disable secure mode

morpheu1988, 2021-05-29:
Hello Community,

after compiling the RSID Viewer with the option RSID_SECURE I'm not able anymore to run the camera with a Viewer without this option enabled.

It gives to me the QueryErrror: Security Error.

I can I disable the secure mode and make my camera working again without it?

Thank you

Regards,

Francesco

alexk1976, 2021-05-29:
once you run security mode once it's paired your host and device. now you need to activate unpair. you use command line tool (rsid-cli.exe) to unpair. After that you can use non-secure mode again

morpheu1988, 2021-05-29:
I had managed to unpair my device forcing and unpair in the source code.
Your suggestion makes everything more smooth. Thank you

Francesco

closed 2021-06-01 by gabime