realsenseai/RealSenseID #77: Can feature vector(in Faceprints) from 0.17.1 be used at 0.18.1?

jeyyoon, 2021-06-01:
Our customer, hundreds people, enrolled their face with 0.17.1.
But I noticed a new version 0.18.1 was released and feature vector at this version has length of 259.
Last three element of the vector is a kind of metadata like wearing a mask or not.

For us, this change can be a critical problem for keeping our rsid having the latest version,
because feature vector length changed and it's worse problem if we can not use feature vector from 0.17.1 anymore at newer version.

So my question is,
- Can I still use database with f.vec from 0.17.1 at 0.18.1? (If I remove last three element)
- Why three more element is added to f.vec? There could be another way to add it as a faceprint class parameters.

alexk1976, 2021-06-01:
Hi,
We are really sorry about it but you will need to enroll all again and I recommend you waiting with it a few more weeks.
3 additional members is a placeholder for future changes database changes so customers won't need to enroll again in case we want to add something to database.
But the main problem is not this placeholder but new recognition model that will be introduced next release and will require re-enrollment.
We plan to finish all database and recognition changes within a few weeks and then it'll be frozen.
Only then makes sense to enroll all people and create final database.
thank you

mdisg, 2021-06-08:
@alexk1976 is there a roadmap of your planned features/availablility available? Would be very good as a customer/developer to know what's planned and more important a schedule when the first stable release is planned.

jeyyoon, 2021-06-11:
@alexk1976 is 0.19.0 version is what you told? I wonder no more need to enroll again even new version is released?

like @mdisg said, when is the stable version going to be released?

alexk1976, 2021-06-12:
Hi,
Unfortunately not yet. The next drop 0.20.0 (in 2 weeks) supposed to be final one regarding database changes

jeyyoon, 2021-06-12:
@alexk1976 
Thank you for the plan. 
Then, is the metric of calculating similarity and feature vector length will be the same both 0.19.0 and 0.20.0?
If it is, we can develop our service in advance.

alexk1976, 2021-06-12:
Yes, feature vector structure, length and match function calculating similarity will be same. But faceprint values inside will be changed due to new NN model.

jeyyoon, 2021-06-27:
@alexk1976 
Hi, could you update the schedule for next version SDK?(0.20.0)
Thank you.

alexk1976, 2021-06-27:
Coming Monday or Tuesday

jeyyoon, 2021-06-29:
@alexk1976 
Thank you for updating sdk. 
I expected there would be explanation that there is no more need to re-enroll when new sdk version coming.
But I couldn't see that those kind of comments.

I just wonder..
Does 'no more need to enroll' mean that people don't have to enroll again because we can enroll people with png image?

Otherwise, can we re-use pre-faceprint database for the new sdk version?

Thank you.

alexk1976, 2021-06-29:
yes, you can assume that after this SDK new re-enrollment wont be required. Further changes will be backward compatible. It's not related to png image enrolment but rather to the fact we completed development of all major features and our NN model is stable enough. 
If you want to be on the safe side..better to capture/save dumps of raw10 images during enrollment though 99% it wont be required. I keep 1% of probability for future re-enrollment just in case of a new un-expected bug.
thanks

closed 2021-07-26 by gabime