realsenseai/RealSenseID #78: Linux rsid-viewer error

zeynepkoyun, 2021-06-01:
@RealSenseCustomerSupport  @alexk1976  @mdisg  "rsid-viewer" file is not created with linux. How can I view?

I managed to compile without preview feature and the generated files are as follows.
After building solution you will find in \build\bin\ two executables:
rsid-cli: Command line interface to RealSenseID.
fw-updater-cli: Firmware update tool.

mdisg, 2021-06-01:
I think it's not supported under linux. In the [CMakeLists.txt](https://github.com/IntelRealSense/RealSenseID/blob/cb7b786ed7d092197ffa7e473d39564c776751c9/tools/CMakeLists.txt#L10) file it is only included if MSVC is available:

```
if(MSVC)
    add_subdirectory(rsid-viewer)
endif()
```

zeynepkoyun, 2021-06-01:
@mdisg Thank you for your answer. Since there is "rsid-viewer" in the "tools" folder in the "CMakeLists.txt" file and the definition is made as follows, I did not add it in the current structure like this:

if(MSVC)    
    string(APPEND CMAKE_CXX_FLAGS " /Zc:__cplusplus /MP")    
endif()
...
if (RSID_TOOLS)
    add_subdirectory(tools)
endif()

gabime, 2021-07-26:
rsid-viewer is a windows only charp app.

closed 2021-07-26 by gabime