realsenseai/RealSenseID #80: How can I record frame in linux with Realsense Id?

zeynepkoyun, 2021-06-03:
Linux üzerinde RealsenseId ile komut satırı arayüzü ile çalışıyorum, bu yüzden oluşturulan çerçeveyi görüntüleyemiyorum, görüntüyü kaydedebilir miyim?

For example, saving the processed frame when "a" is pressed as seen in the image below?
[image]

benlev, 2021-06-07:
Uncomment this section in rsid-cli's main.cc:

```
// Enable this code to enable saving images as ppm files
//
//    std::string filename = "outputimage" +  std::to_string(counter) + ".ppm";
//    FILE* f1 = fopen(filename.c_str(), "wb");
//    fprintf(f1, "P6\n%d %d\n255\n", image.width, image.height);
//    fwrite(image.buffer, 1, image.size, f1);
//    fclose(f1);
//

```

zeynepkoyun, 2021-06-08:
@benlev Thanks for your answer, I'm looking as you stated now, but the commands are written under the "#ifdef RSID_PREVIEW" condition and I tried to remove this condition before, but during compilation, only "rsid-cli" and "rsid-fw-update" files were created on Linux.
Therefore, I could not activate "rsid-preview".
I would be very happy if you could help with this.

3dsf, 2021-06-09:
do you want to use the realSenseID library, or would you be okay to treat it like a camera?  

Here is a poor (slow?) example of how to get a preview image without preview enabled  
`ffmpeg -f video4linux2 -i /dev/video0 -ss 0:0:2 -frames 1 out.jpg -y`  

There are many other ways to snap a frame with linux while treating it as a camera.  In my limited testing, the only time it may fail is if you are asking the device to do another operation at the same time.  Try `ffplay /dev/video0` and then run authentication and other commands.  As of the last firmware I ran the colour space is correct, but the camera will present different colours in full spectrum light (such as sunlight) than you might expect.  It is a minor issue, though may cause a twinge in some users; I believe it's caused due to the sensors reading a wider spectrum than is visible to humans.

zeynepkoyun, 2021-06-09:
Hi @3dsf @benlev 
I was able to see the pictures, but I'm looking to record a single frame instead of multiple frames. I would be very happy if you could help me with this subject.
"c to capture images from device." Can only one image be saved when clicked?

3dsf, 2021-06-10:
I would recommend that you google something like `capture frame from webcam with opencv c++`

zeynepkoyun, 2021-06-10:
> google gibi bir şey yapmanı tavsiye ederim `capture frame from webcam with opencv c++`

@3dsf  Thank you for your answer, When I look at the sample codes I want to learn exactly, frames are taken for 3 seconds for the "c" option. As a temporary solution, I try to change the 3sn value here according to my needs.

closed 2021-07-23 by alexk1976