realsenseai/RealSenseID #81: undefined reference to `RealSenseID::FaceAuthenticator::FaceAuthenticator()

zeynepkoyun, 2021-06-07:
When I try to run enroll.cc I get the error "undefined reference to `RealSenseID::FaceAuthenticator::FaceAuthenticator()".
I would be glad if you help.

lmjieSCU, 2021-06-07:
maybe you can try as this:
./enroll -l rsid -l rsid_c

lmjieSCU, 2021-06-07:
the librsid.so and librsid_c.so file you can find in build dir after running

lmjieSCU, 2021-06-07:
the librsid.so and librsid_c.so file you can find in build dir after running sudo make install

zeynepkoyun, 2021-06-07:
@lmjieSCU Thanks for your answer, the files you mentioned were created when the compilation was complete but when I tried to run the sample code I got this error
Now I'm looking again.

zeynepkoyun, 2021-06-07:
> belki şöyle deneyebilirsiniz:
> ./enroll -l rsid -l rsid_c
Hi @lmjieSCU 
Could I be using the above statement incorrectly? I'm trying to run it using the "g++ enroll.cc" command on Linux 16.04.

benlev, 2021-06-07:
Please try the following:

1. Make sure your device is unpaired.
2. Re-generate the solution with RSID_SECURE=0.
3. Re-compile the enroll.c sample code.

zeynepkoyun, 2021-06-08:
> Please try the following:
> 
> 1. Make sure your device is unpaired.
> 2. Re-generate the solution with RSID_SECURE=0.
> 3. Re-compile the enroll.c sample code.
@benlev Thanks for your answer, I tried to do as per the above steps but I get the same error.

RealSenseSupport, 2021-06-29:
when you build the sample, you may need add some arguments.

> -I dir
>     Add the directory dir to the list of directories to be searched for header files during preprocessing.
> -Ldir
>     Add directory dir to the list of directories to be searched for -l.

https://gcc.gnu.org/onlinedocs/gcc/Directory-Options.html

gabime, 2021-07-26:
@zeynepkoyun Did you managed solve the issue ?

closed 2021-08-23 by zeynepkoyun