realsenseai/RealSenseID #85: Failed connecting to port /dev/ttyACM0 status:Error

zeynepkoyun, 2021-06-10:
When I got this error, it was solved with the command "sudo chmod a+rw /dev/ttyACM0" before, but now the error continues.
I would be glad if you help.

3dsf, 2021-06-10:
you F455 could be at a different port depending on if you have other usb devices (most likely a camera) attached.  
run 
`sudo dmesg | grep 'Intel F450' -A 5 | grep ttyACM`
to see where it is attached  

using 
`sudo usermod -a -G dialout $USER`
is a more robust solution for granting permissions to a user, especially if the port allocation might change

https://github.com/IntelRealSense/RealSenseID/tree/master/tools#linux----compilation-and-usage  

---  

Or you could also look at a solution like https://unix.stackexchange.com/questions/66901/how-to-bind-usb-device-under-a-static-name (and add in permissions similar to how they did with the librealsense cameras after assigning it a static address)

---  

Each method has different permission implications and some may work better or worse (or have no significant affect) for your use case

zeynepkoyun, 2021-06-14:
@3dsf Thank you for your answer,

I ran the commands as you mentioned in your comment. (https://github.com/IntelRealSense/RealSenseID/tree/master/tools#linux----compilation-and-usage)
However, when the error persisted, I saw on a different site.
I tried the command "sudo chmod a+rw /dev/ttyACM0" but when the error occurs, I run the command "sudo dmesg | grep 'Intel F450' -A 5 | grep ttyACM" and multiple devices appear. It fixes itself after a while.

closed 2021-07-26 by gabime