realsenseai/RealSenseID #86: Enroll user with picture

luca18, 2021-06-11:
Would it be possible to configure the SDK or modify the source code so that the camera can enroll a person just with a picture of the given person? Right now, when I try to enroll myself with a picture of myself, it says Face Not Detected or Spoofing.

alexossa1, 2021-06-11:
I think is not possible, the F455 uses the depth camera to extract the face features to compare with the data base

3dsf, 2021-06-11:
I'm not with Intel, but I don't think that is in the plan.  What makes this
device special is the biometric security features.

That being said, there is nothing stopping you from triggering a
realsenseID enrollment action based on face recognition using a different
library.

alexk1976, 2021-06-12:
Hello guys,
The feature is part of our plan and already in development
You will be able to enroll using .png image

3dsf, 2021-06-13:
@alexk1976 
Interesting, I assumed it could not done since there was not a conversion mechanism between different models. https://github.com/IntelRealSense/RealSenseID/issues/77  

Does a match with the png trigger an enrolment action, or will the submission of the png be the enrolment action?

luca18, 2021-06-14:
@alexk1976 
Do you have a timeline for this feature? What would be the approximate release date?

alexk1976, 2021-06-16:
I assume it's be ready end of next week.
The feature will enable you to send us png image for enrollment only.
Authentication still will require real camera only as before and no plan to enable to authenticate using png images.

RealSenseSupport, 2021-07-08:
@luca18 You can try EnrollImage API in RSID v0.21.0
https://github.com/IntelRealSense/RealSenseID/releases/tag/v.21.0

gabime, 2021-07-26:
Supported since version 0.21.0.

closed 2021-07-26 by gabime