realsenseai/RealSenseID #89: Will the library support returning depth image constructed by the F455 device?

pvhduc97, 2021-06-26:
I really want to get the depth image for other operations 😄
Is this feature being developed. If not, can you give us the estimated release time for the feature?

RealSenseSupport, 2021-06-30:
Hi pvhduc97,

F455 does not support to output 'Depth Image'. Thanks!

pvhduc97, 2021-06-30:
Hello @RealSenseSupport, you mean F455 will never support output 'depth image'?

'depth image' : something similar to this:
[image]

RealSenseSupport, 2021-07-02:
Hi pvhduc97,

Thanks for detail samples!
But there is no disparity output from F455 either.

3dsf, 2021-07-02:
@RealSenseCustomerSupport 

After looking at something like [this piece of marketing](https://www.intelrealsense.com/wp-content/uploads/2020/12/RealSense_Digram_1440px_Wide_v5.mp4) — is there access to mapped data points in which a mesh like that could be overlain?

RealSenseSupport, 2021-07-08:
The disparity will be used by internal algorithm only.

pvhduc97, 2021-07-08:
Thank @RealSenseSupport, I got it.
I think due to some security risks for the device, your team cannot provide the disparity map.

gabime, 2021-07-26:
F455 does not support constructing the depth image.

closed 2021-07-26 by gabime