realsenseai/RealSenseID #91: [Feature Request] Linux User ID string limit

id10tt-attnet, 2021-06-28:
The version "rsid_sdk_v0.19.0_signed_win64" allows for a 36 character string for USER ID. The version "RealSenseID-0.19.0" allows for a 30 character string for USER ID.  This confirmed for FW "F450_3.1.0.29_SKU2_SIGNED.bin".

Please standardize on the 36 character namespace.

alexk1976, 2021-06-29:
Our limitation is 30 characters for ID. I dont know where you seen 36?

mdisg, 2021-06-30:
I had no time yet to look into the new release but maybe the firmware was changed/prepared according to the request #53 @alexk1976 ?
36 characters would perfectly fit the string representation of a Guid for example `73A4962D-FD64-4EFC-9956-8B20904B41E3`

alexk1976, 2021-06-30:
I see..so we'll provide it in one of the following releases.

mdisg, 2021-06-30:
That would be nice

id10tt-attnet, 2021-06-30:
Thank you. The linux vers. 0.19.0 _may_ return a 36-char ID upon authorization, however the enrollment input only allows for 30.

This case should be closed.

closed 2021-07-08 by mengyui