realsenseai/RealSenseID #92: can not get faceprints data from ExtratdFaceprints class [Android]

tmq902005, 2021-06-29:
Hi Team!

In AuthFaceprintsExtractionCallback, I can not get FacePrints Data.

[image]

mengyui, 2021-07-08:
It seems it's an swig issue.
You can try this commit as a workaround
https://github.com/mengyui/RealSenseID/commit/ec11944bd340b60962e3669a63122b4d2b1841a0

tmq902005, 2021-07-13:
Thank you @mengyui , can you merge branch android-hostmode-fix to ver 21

mengyui, 2021-08-11:
Try this https://github.com/mengyui/RealSenseID/commit/caad1659c9c5664b44486abb1332561ba6190d8b. rebase on v0.24.0

mcostanzo, 2021-08-23:
I'm having the same issue, is this going to be fixed in the main repo?

mcostanzo, 2021-08-23:
@gabime @Gilaadb @RealSenseSupport could you check on this?