realsenseai/RealSenseID #93: Unable to start session on Ubuntu 18.04 with Release 0.21.0

jvettese, 2021-06-29:
I attempted to build and run the rsid-cli as follows

`gcc -g main.cc /RealSenseID-v.21.0/build/lib/librsid.so -lstdc++ -o main -I/RealSenseID-v.21.0/include`
 
When I tried to run:
`user@host:/RealSenseID-v.21.0/tools/rsid-cli$ ./main /dev/ttyACM0
[2021-06-29 08:30:25.133] [debug] [LinuxSerial] Opening serial port /dev/ttyACM0 baudrate 115200
Connected to device
[2021-06-29 08:30:25.252] [debug] [NonSecureSession] Start session
[2021-06-29 08:30:25.252] [debug] [PacketSender] Sending packet 'o'
[2021-06-29 08:30:25.253] [debug] [PacketSender] Waiting packet..
[2021-06-29 08:30:30.355] [error] [NonSecureSession] Failed to recv device start session response
[2021-06-29 08:30:30.355] [error] [FaceAuthenticatorImpl] Session start failed with status 103
Status: SerialError

[2021-06-29 08:30:30.355] [debug] [NonSecureSession] Close session
Please select an option:

  'e' to enroll.
  'a' to authenticate.
  'd' to delete all users.
  's' to set authentication settings.
  'g' to query authentication settings.
  'u' to query ids of users.
  'n' to query number of users.
  'b' to save device's database before standby.
  'v' to view additional information.
  'x' to ping the device.
  'q' to quit.
 
server mode options:
  'E' to enroll with faceprints.
  'A' to authenticate with faceprints.
  'U' to list enrolled users
  'D' to delete all users.

> E
User id to enroll: c
[2021-06-29 08:30:41.670] [debug] [LinuxSerial] Opening serial port /dev/ttyACM0 baudrate 115200
[2021-06-29 08:30:41.670] [error] [FaceAuthenticatorImpl] Failed open serial port. errno: 2
Failed connecting to port /dev/ttyACM0 status:Error
`

I did ensure that /dev/ttyACM0 had proper privileges, and that the camera was on the correct firmware.

alexk1976, 2021-06-29:
what FW u have on device? did you install F450_4.0.0.37_SKU1_SIGNED.bin FW?

benlev, 2021-06-29:
Hi @jvettese,
1. Which flags did you use to generate the solution and compile?
2. Using 'rsid-viewer' on Windows - are you able to Enroll/Authenticate normally?

jvettese, 2021-06-29:
@benlev 

1. I am compiling with:  `gcc -g main.cc /RealSenseID-v.21.0/build/lib/librsid.so -lstdc++ -o main -I/RealSenseID-v.21.0/include`

2.  I am able to Enroll/Authenticate normally. 

Also I am able to use the 'rsid-cli' on Windows without issue. When I try to build a slightly modified version I get this error:
`[2021-06-29 09:51:18.988] [error] [WindowsSerial] Error while reading from serial port. Last error: 16` 
I am not sure if this is related to my issue with linux

benlev, 2021-06-30:
@jvettese,
Please try the following to see if it solves the issue:
``` sudo systemctl stop ModemManager ```

jvettese, 2021-06-30:
This was solved by reconfiguring LD_LIBRARY_PATH

closed 2021-06-30 by jvettese