realsenseai/RealSenseID #94: Error with Release 0.21.0

luca18, 2021-06-29:
I compiled the code from source on Ubuntu 18.04. I checked that the serial number matches SKU1 and updated the firmware to F450_4.0.0.37_SKU1_SIGNED.bin.

I get the following error when I run ./rsid-cli /dev/ttyACM0 usb

[2021-06-29 16:03:15.769] [debug] [PacketSender] Received packet 'Y' after 167 millis
[2021-06-29 16:03:15.770] [error] [NonSecureSession] Failed to recv device start session response
[2021-06-29 16:03:15.772] [debug] [PacketSender] Sending packet 'q'
[2021-06-29 16:03:15.773] [debug] [PacketSender] Waiting packet..
[2021-06-29 16:03:15.791] [debug] [PacketSender] Received packet 'Y' after 17 millis
[2021-06-29 16:03:15.791] [error] [NonSecureSession] Invalid sequence number. Last: 0, Current: 0
[2021-06-29 16:03:15.791] [error] [FaceAuthenticatorImpl] Failed receiving fa packet (status 106)
Status: SerialSecurityError

alexk1976, 2021-06-30:
benlev commented 4 hours ago:
Please try the following to see if it solves the issue:
sudo systemctl stop ModemManager

luca18, 2021-06-30:
@jvettese 
How do I ensure that I am not being linked to the wrong librsid.so? Where is the correct librsid.so and how did you link it to your LD_LIBRARY_PATH env variable?

jvettese, 2021-06-30:
`export LD_LIBRARY_PATH={PATH TO RSID LIB DIR}`
I have this set in my ~/.bashrc

You could also do `sudo find / -name librsid.so` to see where all librsid.so live

luca18, 2021-06-30:
I tried both suggestions:

1.
systemctl stop ModemManager

2.
echo "export LD_LIBRARY_PATH=/home/luca/Projects/RealSenseID/build/lib/" >> ~/.bashrc
source .bashrc

No luck.

benlev, 2021-06-30:
Try re-generating the solution using RSID_SECURE=1 and re-run the sample - does it solve the issue?

luca18, 2021-06-30:
After compiling with RSID_SECURE=1 and running ./rsid-cli /dev/ttyACM0 usb, I get the following error:

./rsid-cli: symbol lookup error: /home/luca/Projects/RealSenseID-Secure/build/lib/librsid_secure_helper.so: undefined symbol: mbedtls_hmac_drbg_random

This seems worse since I don't even get the prompt.

luca18, 2021-06-30:
I've also noticed something that seems odd. When I compile with cmake .., I see this at the end:

-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/luca/Projects/RealSenseID-Secure/build

Is it normal that it says "not found" for two of the threads?

3dsf, 2021-07-09:
@luca18 , did you get it working?

---  

you are using linux, so this is normal 

> -- Looking for pthread_create - not found
> -- Looking for pthread_create in pthreads
> -- Looking for pthread_create in pthreads - not found
> -- Looking for pthread_create in pthread
> -- Looking for pthread_create in pthread - found

It checks for the non-linux first and then finds it when it checks for the linux location.  

---   

do this to disable modem manager (as per @alexk1976's advice)  *I've disabled modem manager for usb tethering reasons and I've never had an issue like you've experienced.*
- sudo systemctl stop ModemManager
- sudo systemctl disable ModemManager 

---  

I've had a suspicion that you have more than one usb peripheral that uses a communication port and you might not be at `/dev/ttyACM0` 
Confirm where it is connected by running 
`sudo dmesg | grep 'Intel F450' -A 5 | grep ttyACM`

luca18, 2021-07-14:
I've tried everything that was suggested and nothing seems to work.
I've also tried to run it on multiple ubuntu machines and a windows computer with no success whatsoever.

I thinking there might be a problem with the F450 camera that I have. It was working fine before the new firmware update.
Is it possible to do a factory reset on the camera or downgrade to the old firmware?

gabime, 2021-07-26:
Please try latest 0.22.0 version with the new FW and see if it solves your problem.

luca18, 2021-07-26:
I solved the issue by following the solution posted at #47.

closed 2021-07-26 by luca18