realsenseai/RealSenseID #95: Error with install

ajtudela, 2021-07-07:
I found some errors trying to configure cmake with install option: cmake .. -DRSID_INSTALL=ON

```
-- Build type: Release
-- Linux platform
-- CMAKE_BUILD_TYPE = Release
-- VERSION = 2.1.0, BUILD = 20210707
-- 64-bit build (x86_64)
-- CMAKE_INSTALL_PREFIX = /opt/RealSenseID
-- CMAKE_INSTALL_BINDIR = bin (/opt/RealSenseID/bin)
-- CMAKE_INSTALL_DATAROOTDIR =  (/opt/RealSenseID)
-- CMAKE_INSTALL_DOCDIR = doc (/opt/RealSenseID/doc)
-- CMAKE_INSTALL_INCLUDEDIR = include (/opt/RealSenseID/include)
-- CMAKE_INSTALL_LIBDIR = lib64 (/opt/RealSenseID/lib64)
-- CMAKE_INSTALL_MANDIR = man (/opt/RealSenseID/man)
-- Shared libraries disabled (ENABLE_SHARED = 0)
-- Static libraries enabled (ENABLE_STATIC = 1)
-- 12-bit JPEG support disabled (WITH_12BIT = 0)
-- Arithmetic decoding support enabled (WITH_ARITH_DEC = 1)
-- Arithmetic encoding support enabled (WITH_ARITH_ENC = 1)
-- TurboJPEG API library disabled (WITH_TURBOJPEG = 0)
-- TurboJPEG Java wrapper disabled (WITH_JAVA = 0)
-- In-memory source/destination managers enabled (WITH_MEM_SRCDST = 1)
-- Emulating libjpeg API/ABI v6.2 (WITH_JPEG7 = 0, WITH_JPEG8 = 0)
-- libjpeg API shared library version = 62.3.0
-- Compiler flags =  -O3 -DNDEBUG
-- Linker flags =  
-- Compiler supports pointers to undefined structures.
-- INLINE = __inline__ __attribute__((always_inline)) (FORCE_INLINE = 1)
-- Linker supports GNU-style version scripts
-- CMAKE_EXECUTABLE_SUFFIX = 
-- SIMD extensions: None (WITH_SIMD = 0)
-- FLOATTEST = no-fp-contract
-- RPM architecture = x86_64, DEB architecture = amd64
-- Generating install..
CMake Error at cmake/Install.cmake:19 (install):
  install TARGETS given target "rsid-cli" which does not exist in this
  directory.
Call Stack (most recent call first):
  CMakeLists.txt:119 (include)

CMake Error at cmake/Install.cmake:26 (install):
  install TARGETS given target "rsid" which does not exist in this directory.
Call Stack (most recent call first):
  CMakeLists.txt:119 (include)

-- Configuring incomplete, errors occurred!
```

3dsf, 2021-07-07:
To me it looks like you tried to install to /opt/ without sufficient permissions...

ajtudela, 2021-07-07:
I just did `cmake ...` not `make install`

3dsf, 2021-07-08:
> -- CMAKE_INSTALL_DATAROOTDIR =  (/opt/RealSenseID)

My understanding is it's a configuration problem with your cmake install.  It looks like you are running a version of cmake which is trying to populate the build files in the `/opt` directory.  When you run **cmake**, it makes files for **make** to use.  `/opt/` is a great place to keep 3rd party programs that are installed outside of your package manager.

closed 2021-07-26 by gabime