realsenseai/RealSenseID #96: SecurityError

luca18, 2021-07-08:
Hi,

I keep on getting the same error when I start the RealSense ID Viewer Software on Windows 10. The error persists when I try to enroll or authenticate. 

I've updated the firmware to the most recent sku1 version. I think there is a problem with my camera since it's giving me errors everywhere Windows or Ubuntu regardless of everything I try.

Is there a way to hit a 'factory reset' for the camera?

[image]

lmjieSCU, 2021-07-26:
I got it as well. Have you solved it?Thanks very much

gabime, 2021-07-26:
Please see https://github.com/IntelRealSense/RealSenseID/issues/47 on possible solution.

luca18, 2021-07-26:
Thank you @gabime for the reference.
This solved the problem.

closed 2021-07-26 by luca18