realsenseai/RealSenseID #97: enroll error with 0.21.0

donghyun999, 2021-07-13:
When registering a face If you don't rotate your face and just stare at the camera, it won't be a problem.
When rotating the face, the error occurs as shown in the photo below.

I compiled the code from source on window10, I checked that the serial number matches SKU1 and updated the firmware to F450_4.0.0.37_SKU1_SIGNED.bin.

[image]

JR-Malin, 2021-07-16:
I am working with custom C#/.NET code on Windows 10. The latest version was running successfully on SDK V 0.18.1. I just installed and updated my code for SDK V 0.21.0.  I'm using the F455 SKU1 hardware and the firmware has been updated to V4.0.0.37 SKU1.  I am seeing the same problem with enrollment using my custom software as has been reported here.  As soon as I move my head, during the enrollment operation, the SKD returns an error of either faceprint not found or spoof.  If I hold my head very still at about 4 inches away from the sensor, I can get a successful enrollment, but it it difficult.  Authentication seems to work. I thought I had a bug in my code adaptation to the new faceprint structures, but after seeing this post, I tested the RealSense ID Viewer, and I see the same problem with that software. I have not seen a response to this post nor any other comments verifying the same problem. I can confirm that this is not a one-off. I hope you will look into what has changed in V 0.21.0 and post a solution.

agodlin, 2021-07-19:
@donghyun999 
Thank for your issue report, this issue will be fixed in the next release. 
for now, if you make sure that the first frame of enrollment contain a frontal face, it should work perfectly.

alexk1976, 2021-07-19:
next release is expected end of this week

gabime, 2021-07-26:
Fixed in release 0.22.0

closed 2021-07-26 by gabime