realsenseai/RealSenseID #99: Cannot connect F455 with Intel Realsense ID Viewer

PavKovalenko, 2021-07-17:
On previous versions camera works fine. But after update cannot connect camera

In update process device throw error, And it happens each update time.
And currently when connect camera to Windows, Intel RealSense display 
"Device Error"
Device failed to respond
Intel RealSense ID v0.212.0
Firmware - 4.0.0.37

In console:
 [WindowsSerial] Opening serial port COM9
 [debug] [PacketSender] Sending packet 'p'
 [debug] [PacketSender] Waiting packet..
 [error] [PacketSender] Protocol version doesn't match. Expected: 2, Received: 112
[error] [DeviceControllerImpl] Failed receiving data packet (status 107)
[debug] [WindowsSerial] Closing serial port

If I connect to Android device error message is "Supported USB device not found"
Android SDK 0.21.0

Please help. How to connect camera
[image]

RealSenseSupport, 2021-07-20:
Hi PavKovalenko,
It seems there is a failure when upgrade FW.
Could you share the serial number of the F455?

I guess you can try to re-flash/re-write the FW using <Intel RealSenseID SDK>\tools\rsid-fw-update.exe
or try to downgrade FW to 3.1.x and then upgrade to 4.0.x again.

PavKovalenko, 2021-07-20:
Serial number of device: S/N: 10446228095599A27FM10464-100
When I tried to downgrade to 3.1 it downgraded, but after updating I see error 'QueryDeviceConfig Error - Failed to query device config'
And no camera preview.
Same error when I tried to update to 4.0 and no preview

Could you please clarify how I can  re-flash/re-write the FW using \tools\rsid-fw-update.exe. Because currently in tools folder no such .exe file only "rsid-fw-update" folder with C++ file

RealSenseSupport, 2021-08-11:
if you have install the pre-built sdk package (for example download from https://github.com/IntelRealSense/RealSenseID/releases/tag/v0.24.0), you can find the tool in C:\Program Files\Intel RealSenseID SDK\tool\rsid-fw-update.exe

Cryptohash, 2023-04-20:
I'm having the same issue. What this ever resolved?