realsenseai/RealSenseID #100: C# wrapper x86

jbres4, 2021-07-22:
Hi,
I made C# development based on the C# wrapper included on the SDK. It's working fine.
I was wondering if it would be possible to have a x86 version of the wrapper instead of the current x64 ? We have currently an existing project, which already uses different libraries all compiled in x86 (because one them come from third parties and only x86 versions are available). So to include our implementation of your C# wrapper in our project, we would need a x86 version.
Thanks.

gabime, 2021-07-26:
Hi,
It is not officially supported, but you can try the following  steps:

1. Comment out the ```set(CMAKE_CSharp_FLAGS "/platform:x64")``` from the wrappers/csharp/CMakelists.txt file
2.  cmake -G "Visual Studio 16 2019" -A Win32  -DRSID_PREVIEW=1

closed 2021-07-26 by gabime

jbres4, 2021-07-27:
Hi,
I've downloaded the latest release 0.23.0 today and tried what you suggested:

- Install CMake 3.21 (Visual Studio 2019 already installed)
- Comment (or remove) `set(CMAKE_CSharp_FLAGS "/platform:x64")` from the `wrappers/csharp/CMakeLists.txt` file
- Run `cmake -G "Visual Studio 16 2019" -S "<source_path>" -B "<source_path>\build" -A Win32 -DRSID_PREVIEW=1`
- Open with VS2019 file `build\RealSenseID.sln` created by CMake
- Select `Release` configuration and build

Then I tried to use these dll in my sample app (compiled in x86 in this case) which basically:

- Display the preview of the camera
- Enroll a user with an image with `Authenticator.EnrollImage` method
- Authenticate a user with `Authenticator.Authenticate` method

Preview, and enroll are working fine. But when I try to use `Authenticator.Authenticate` method, I see the red light of the camera and just after I get a `System.AccessViolationException`. StackTrace :
```
   at rsid.Authenticator.rsid_authenticate(IntPtr rsid_authenticator, AuthArgs& authArgs)
   at rsid.Authenticator.Authenticate(AuthArgs args)
```

Below is the CMake command output if it helps
```
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19042.
-- The C compiler identification is MSVC 19.29.30040.0
-- The CXX compiler identification is MSVC 19.29.30040.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/VC/Tools/MSVC/14.29.30037/bin/Hostx64/x86/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/VC/Tools/MSVC/14.29.30037/bin/Hostx64/x86/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type: Release
-- Windows platform
-- CMAKE_BUILD_TYPE = Release
-- VERSION = 2.1.0, BUILD = 20210727
-- 32-bit build (i386)
-- CMAKE_INSTALL_PREFIX = c:/RealSenseID
-- CMAKE_INSTALL_BINDIR = bin (c:/RealSenseID/bin)
-- CMAKE_INSTALL_DATAROOTDIR =  (c:/RealSenseID)
-- CMAKE_INSTALL_DOCDIR = doc (c:/RealSenseID/doc)
-- CMAKE_INSTALL_INCLUDEDIR = include (c:/RealSenseID/include)
-- CMAKE_INSTALL_LIBDIR = lib (c:/RealSenseID/lib)
-- Shared libraries disabled (ENABLE_SHARED = 0)
-- Static libraries enabled (ENABLE_STATIC = 1)
-- 12-bit JPEG support disabled (WITH_12BIT = 0)
-- Arithmetic decoding support enabled (WITH_ARITH_DEC = 1)
-- Arithmetic encoding support enabled (WITH_ARITH_ENC = 1)
-- TurboJPEG API library disabled (WITH_TURBOJPEG = 0)
-- TurboJPEG Java wrapper disabled (WITH_JAVA = 0)
-- In-memory source/destination managers enabled (WITH_MEM_SRCDST = 1)
-- Emulating libjpeg API/ABI v6.2 (WITH_JPEG7 = 0, WITH_JPEG8 = 0)
-- libjpeg API shared library version = 62.3.0
-- Compiler flags = /DWIN32 /D_WINDOWS /W3 /W3 /wd4996 /MD /O2 /Ob2 /DNDEBUG
-- Linker flags = /machine:X86 /INCREMENTAL:NO
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of size_t
-- Check size of size_t - done
-- Check size of unsigned long
-- Check size of unsigned long - done
-- Performing Test HAVE_BUILTIN_CTZL
-- Performing Test HAVE_BUILTIN_CTZL - Failed
-- Looking for include file intrin.h
-- Looking for include file intrin.h - found
-- Performing Test INLINE_WORKS
-- Performing Test INLINE_WORKS - Success
-- INLINE = __forceinline (FORCE_INLINE = 1)
-- CMAKE_EXECUTABLE_SUFFIX = .exe
-- SIMD extensions: None (WITH_SIMD = 0)
-- FLOATTEST = msvc
-- The CSharp compiler identification is Microsoft unknown 9999
-- The CSharp compiler version is 9999
-- Check for working C# compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/MSBuild/Current/Bin/Roslyn/csc.exe
-- Check for working C# compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/MSBuild/Current/Bin/Roslyn/csc.exe - works
-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Sources/Tests/RealSenseID-v.23.0 x86/build
```

The sample app is working fine when compiled in x64 and is using the released dll, or if if create the dll using CMake and default configuration.

Do you know what may cause this issue ?

Thanks.

gabime, 2021-07-27:
Hard to tell.  Like I said it is not supported. If you find a solution a PR would be welcome!

luisruizlatic, 2022-04-19:
Hi @jbres4, I was wondering if you were able to use the SDK in x86, that would help us a lot!

jbres4, 2022-04-25:
Hi @luisruizlatic, no I was not able to have it work. And as the project is dead (the camera F455 is not available anymore since February 2022), I stopped looking at this...