realsenseai/RealSenseID #105: RuntimeError: CrcError

zeynepkoyun, 2021-07-28:
I have compiled the RealsenseID to work with python3.6 and run "viewer.py" but when I want to run "authenticate.py" I get the following error.
@gabime  I would be glad if you help.

[2021-07-28 15:46:42.814] [debug] [LinuxSerial] Opening serial port /dev/ttyACM0 baudrate 115200
[2021-07-28 15:46:42.931] [debug] [NonSecureSession] Start session
[2021-07-28 15:46:42.931] [debug] [PacketSender] Sending packet 'o'
[2021-07-28 15:46:42.933] [debug] [PacketSender] Waiting packet..
[2021-07-28 15:46:42.941] [error] [PacketSender] Got invalid crc. Expected: 561. Actual: 20100
[2021-07-28 15:46:42.941] [error] [NonSecureSession] Failed to recv device start session response
[2021-07-28 15:46:42.941] [error] [FaceAuthenticatorImpl] Session start failed with status 108
on_result AuthenticateStatus.CrcError
Traceback (most recent call last):
  File "authenticate.py", line 22, in <module>
    f.authenticate(on_faces=on_faces, on_result=on_result)
RuntimeError: CrcError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "authenticate.py", line 22, in <module>
    f.authenticate(on_faces=on_faces, on_result=on_result)
TypeError: __exit__(): incompatible function arguments. The following argument types are supported:
    1. (self: rsid_py.FaceAuthenticator, arg0: capsule, arg1: capsule, arg2: capsule) -> None

Invoked with: <rsid_py.FaceAuthenticator object at 0x7f3df1b56ae8>, <class 'RuntimeError'>, RuntimeError('CrcError',), <traceback object at 0x7f3df1b6d788>
[2021-07-28 15:46:42.946] [debug] [NonSecureSession] Close session

gabime, 2021-07-28:
Please post a sample code to reproduce

gabime, 2021-07-28:
Just to clarify, the viewer.py worked, but the authenticate.py didn’t? both in linux?

jeyyoon, 2021-07-28:
I had the problem and solved it by upgrading rsid firmware with the latest version.

zeynepkoyun, 2021-07-28:
@gabime yes, both on linux

zeynepkoyun, 2021-07-28:
> I had the problem and solved it by upgrading rsid firmware with the latest version.

@jeyyoon I tried "F450_4.2.0.8200_SKU2_SIGNED.bin" and "F450_4.2.0.8200_SKU1_SIGNED.bin" but got the same error.

zeynepkoyun, 2021-07-28:
@gabime I'm getting the latest version from the site "https://github.com/IntelRealSense/RealSenseID/releases/tag/v0.23.0".

gabime, 2021-07-29:
I can't reproduce. Could be that ModemManager is running. Please try

```
sudo systemctl stop ModemManager
```

zeynepkoyun, 2021-07-29:
> I can't reproduce. Could be that ModemManager is running. Please try
> 
> ```
> sudo systemctl stop ModemManager
> ```

@gabime, @jeyyoon  I checked, the service was already not running. I started retesting everything and when I want to update with file "F450_4.2.0.8200_SKU2_SIGNED.bin" it says "Device does not support the encryption applied on the firmware. Replace firmware binary." i got the error. I am working with Ubuntu 16.04.

benlev, 2021-07-29:
Sounds like you have a SKU1 device, please try flashing [this firmware].(https://github.com/IntelRealSense/RealSenseID/releases/download/v0.23.0/F450_4.2.0.8200_SKU1_SIGNED.bin)

zeynepkoyun, 2021-07-29:
> Bir SKU1 cihazınız var gibi görünüyor, lütfen [bu bellenimi güncellemeyi](https://github.com/IntelRealSense/RealSenseID/releases/download/v0.23.0/F450_4.2.0.8200_SKU1_SIGNED.bin) deneyin .

@benlev, I was able to update with this file before but I get CrcError error in "auhenticate" operation so I tried different .bin files but it didn't work.

zeynepkoyun, 2021-07-29:
> > I had the problem and solved it by upgrading rsid firmware with the latest version.
> 
> @jeyyoon I tried "F450_4.2.0.8200_SKU2_SIGNED.bin" and "F450_4.2.0.8200_SKU1_SIGNED.bin" but got the same error.
@jeyyoon, @gabime, I do the firmware update command as "./rsid-fw-update --file F450_4.2.0.8200_SKU1_SIGNED.bin --port /dev/ttyACM0". Are you doing something different in the update part?

closed 2021-08-03 by zeynepkoyun