realsenseai/RealSenseID #110: Now camera Preview

mdisg, 2021-08-03:
I have one machine that I'm working on which does not get the camera preview. The rsid-viewer application does connect to the unit but the preview is empty. The camera is showing up in the device manager.

What can I check? Is there a windows setting interfering?

Monica2511, 2021-08-06:
You should check the version of firmware, i think you are using incompatible  firmware

mdisg, 2021-08-09:
@Monica2511 the same unit works on multiple other PCs so I don't think its the firmware itself it must be something on that one machine.

mdisg, 2021-08-17:
I did another try in debugger and got the following error:

`[error] [Preview] Streaming ERROR : init MSMF backendMSMF failed with  HResult error: 80070005`

mdisg, 2021-08-23:
Issue can be closed. Windows app settings blocked the camera.