realsenseai/RealSenseID #111: "image.get_buffer" Show Old Images Error

zeynepkoyun, 2021-08-03:
Hi @gabime 

When I try to get the image with image.get_buffer, it first takes the old data and then shows the new data.
How can I clear the cache?

gabime, 2021-08-03:
Not sure I understand. Please provide a sample code to reproduce the problem.

zeynepkoyun, 2021-08-03:
Hi @gabime 
When I want to ask for the Anti-Spoof answer on the image I got from the camera, it works asynchronously because it takes the images in order.

That is, while taking the 1st frame and looking for the anti-spoof response, the 2nd, 3rd and 4th frames should be taken. When I asked for the frame for the anti-spoof response again, it should have asked for the 4th frame, but first the 2nd, then the 3rd frame and then the 3rd frame. There is the latest 4thframe answer.