realsenseai/RealSenseID #114: Building python wrapper

openlayers-droid, 2021-08-04:
hi,
so I followed instructions on building python wrapper, but I couldn't find either rsid_py.pyd nor rsid.ddl after the cmake process. Kindly assist.

Many thanks

ryan185, 2021-08-05:
check in build folder inside lib

openlayers-droid, 2021-08-05:
Many thanks @ryan185 . There were few things that had to be taken care of:

1. not familiar how cmake works, i thought it would just create the python pyd file automatically. instead, i had to run the solution file inside Wrappere->python and build to generate the needed files. 
2. i create an environment with Python 3.8. The python code samples produced an error. Upgraded to python39 and samples work as intended. 
3.  to be able to import rsid_py, other extra steps are needed such as creating __init__.py / adding the folder which hosts the pyd to PYTHONPATH.

i wasted a day figuring this out. So hope this helps someone.

ryan185, 2021-08-06:
Python files are working in windows the sample file?

openlayers-droid, 2021-08-06:
@ryan185 yes they are working after the conducting the steps above.

ryan185, 2021-08-06:
please help in 3 point i am not able to do it and error is ModuleNotFoundError: No module named 'rsid_py'

ryan185, 2021-08-06:
File "C:\RealSenseID-master\samples\python\viewer.py", line 237, in <module>
    main()
  File "C:\RealSenseID-master\samples\python\viewer.py", line 233, in main
    input_loop()
  File "C:\RealSenseID-master\samples\python\viewer.py", line 126, in input_loop
    cmd_exec.get(chr(keycode), lambda: None)()
  File "C:\RealSenseID-master\samples\python\viewer.py", line 89, in enroll_example
    with rsid_py.FaceAuthenticator(PORT) as f:
RuntimeError: Error

openlayers-droid, 2021-08-06:
This happened to me at first as well.  it turns out to be Python version. i was using python 3.9. downgraded to python  3.8 and it worked.

RealSenseSupport, 2021-10-18:
@sbekbulaeva 
The steps like this.
1. mkdir build in the root of this project. (<RealSenseID>/build)
2. run "cmake -DRSID_PY=ON -DRSID_PREVIEW=ON .. " in the build folder
3. run make

rukshanfonseka, 2022-07-08:
Has anyone been successful building the python wrappers in ubuntu/linux? I created an issue here https://github.com/IntelRealSense/RealSenseID/issues/147

rukshanfonseka, 2022-07-11:
> 

Can you explain this? I think this is the same issue I am facing.

rukshanfonseka, 2022-07-11:
> Many thanks @ryan185 . There were few things that had to be taken care of:
>
> 1. not familiar how cmake works, i thought it would just create the python pyd file automatically. instead, i had to run [...]

Can you explain how you managed to overcome issue 3)? I am facing the same problem.

VolandSZ, 2022-07-16:
> @sbekbulaeva The steps like this.
> 
> 1. mkdir build in the root of this project. (/build)
> 2. run "cmake -DRSID_PY=ON -DRSID_PREVIEW=ON .. " in the build folder
> 3. run make

Still can not compile wrapper files for python. OS Windows 10 x64

1 . **mkdir build in the root of this project. (/build)**
root of the project consists of these dirs (my path to project is "f:\Cmake_test\test3\"):

```
f:\Cmake_test\test3\3rdparty\
f:\Cmake_test\test3\cmake\
f:\Cmake_test\test3\docs\
f:\Cmake_test\test3\include\
f:\Cmake_test\test3\LicenseNotices\
f:\Cmake_test\test3\samples\
f:\Cmake_test\test3\scripts\
f:\Cmake_test\test3\src\
f:\Cmake_test\test3\tools\
f:\Cmake_test\test3\wrappers\

```
so i just need to make dir
`f:\Cmake_test\test3\build\`
right?

2. **run "cmake -DRSID_PY=ON -DRSID_PREVIEW=ON .. " in the build folder**
I need to run this command from inside folder `f:\Cmake_test\test3\build\` ?

And in it means that it will generate files needed to MAKE Python wrapper in this folder 
according to "CMakeLists.txt" file located in project root folder `f:\Cmake_test\test3\`

But i get this error:
```
CMake Error at tools/rsid-viewer/CMakeLists.txt:4 (project):
  No CMAKE_CSharp_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "F:/Cmake_test/test3/build/CMakeFiles/CMakeOutput.log".
See also "F:/Cmake_test/test3/build/CMakeFiles/CMakeError.log".
```
Ok i dont really care about CSharp BUT on to the step 3 of your advice

**3. run make**
Since i am on Windows i use MSBuild (17.2.1.25201)

Runing it in `f:\Cmake_test\test3\build\` returns this error:
MSBUILD : error MSB1003 - meaning that this folder doesnt have project files to build

And indeed it doesnt have anything like "sln" or "vcxproj" files.

Such files are located in `f:\Cmake_test\test3\build\CMakeFiles\3.24.0-rc3\` and its subfolders but running MSBuild there doesnt give me `rsid.dll` and `rsid_py*.pyd` files.

MSBuild works fine. It can be seen in `CMakeOutput.log` - `Run Build Command(s):C:/Program Files/Microsoft Visual Studio/2022/Community/MSBuild/Current/Bin/amd64/MSBuild.exe cmTC_c79d6.vcxproj /p:Configuration=Debug /p:Platform=x64 /p:VisualStudioVersion=17.0 /v:m && Microsoft (R) Build Engine версии 17.2.1+52cd2da31 для .NET Framework` (for example)

So the main question - **what am i doing wrong?**