realsenseai/RealSenseID #115: anyone achieve successfully run enroll.py and other python files please help i am wokring in python

ryan185, 2021-08-05:


gabime, 2021-08-05:
Please provide more info on the problem

ryan185, 2021-08-06:
When running viewer.py it is showing camera feed but after pressing key "a", "e", "d" it is terminating
Version: 0.23.0
[2021-08-02 15:20:13.369] [debug] [Utilities] detected capture device.
[2021-08-02 15:20:13.471] [debug] [Preview] Preview started!
[2021-08-02 15:28:18.172] [debug] [WindowsSerial] Opening serial port COM9
[2021-08-02 15:28:18.173] [error] [FaceAuthenticatorImpl] Failed to open serial port. GetLastError: 2
[2021-08-02 15:28:18.174] [debug] [NonSecureSession] Close session
Traceback (most recent call last):
File "viewer.py", line 238, in
main()
File "viewer.py", line 234, in main
input_loop()
File "viewer.py", line 127, in input_loop
cmd_exec.get(chr(keycode), lambda: None)()
File "viewer.py", line 83, in auth_example
with rsid_py.FaceAuthenticator(PORT) as f:
RuntimeError: Error

Monica2511, 2021-08-06:
I think you should check serial port of camera. In samples code, default PORT='COM9'. If u using Windows, you can identify serial port of connected device in Device Manager.

ryan185, 2021-08-16:
[2021-08-16 10:39:08.992] [debug] [WindowsSerial] Opening serial port COM3
User id to enroll: 12
[2021-08-16 10:39:12.742] [debug] [NonSecureSession] Start session
[2021-08-16 10:39:12.743] [debug] [PacketSender] Sending packet 'o'
[2021-08-16 10:39:12.747] [debug] [PacketSender] Waiting packet..
[2021-08-16 10:39:12.753] [error] [PacketSender] Got invalid crc. Expected: 561. Actual: 20100
[2021-08-16 10:39:12.754] [error] [NonSecureSession] Failed to recv device start session response
[2021-08-16 10:39:12.754] [error] [FaceAuthenticatorImpl] Session start failed with status 108
on_result EnrollStatus.CrcError
Traceback (most recent call last):
  File "enroll.py", line 27, in <module>
    f.enroll(user_id=user_id, on_hint=on_hint, on_progress=on_progress, on_faces=on_faces, on_result=on_result)
RuntimeError: CrcError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "enroll.py", line 31, in <module>
    print('Users: ', users)
TypeError: __exit__(): incompatible function arguments. The following argument types are supported:
    1. (self: rsid_py.FaceAuthenticator, arg0: capsule, arg1: capsule, arg2: capsule) -> None

Invoked with: <rsid_py.FaceAuthenticator object at 0x000002D0891FB2F0>, <class 'RuntimeError'>, RuntimeError('CrcError'), <traceback object at 0x000002D0892194C0>
[2021-08-16 10:39:12.761] [debug] [NonSecureSession] Close session
[2021-08-16 10:39:12.765] [debug] [WindowsSerial] Closing serial port

ryan185, 2021-08-16:
any idea for CRC error